gbionics/walking-controllers #4: Do not link the pthread library directly

traversaro, 2018-06-13:
In https://github.com/robotology/capture-point-walking/blob/31eaef8c36419550f25ac043725ecef541b21027/code/cpp/Walking_module/CMakeLists.txt#L100 the `pthread` library is linked directly. If link `pthread` is necessary (I fail to see why, but it possible that is an undeclared public dependency of some of the dependencies, the FindThreads CMake module https://cmake.org/cmake/help/v3.5/module/FindThreads.html should be used instead, to remain compatible with Windows or in general platforms without `pthread` .

GiulioRomualdi, 2018-06-13:
`pthread` seems necessary to use `std::thread`
https://github.com/robotology/capture-point-walking/blob/31eaef8c36419550f25ac043725ecef541b21027/code/cpp/Walking_module/include/TrajectoryGenerator.hpp#L48

Maybe I can use [`yarp::os::Thread`](http://www.yarp.it/classyarp_1_1os_1_1Thread.html) instead of `std::Thread`

traversaro, 2018-06-13:
`std::thread` is ok, no problem. Just add:
~~~
find_package(Threads REQUIRED)
target_link_libraries(${EXE_TARGET_NAME} ... Threads::Threads)
~~~

And it should work fine.

closed 2018-06-16 by GiulioRomualdi