gbionics/walking-controllers #11: Compilation fails with clang due to use of "auto" in method declaration

S-Dafarra, 2018-07-13:
When trying to add the walking related stuff to the [robotology-superbuild](https://github.com/robotology/robotology-superbuild/pull/87) compilation on travis fails due to the following error:

```
In file included from /home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/src/main.cpp:14:
/home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingModule.hpp:283:20: error: 
      'auto' not allowed in function prototype
    bool solveQPIK(auto& solver, const iDynTree::Position& desiredCoMPosition,
                   ^~~~
/home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingModule.hpp:377:10: warning: 
      'onTheFlyStartWalking' overrides a member function but is not marked
      'override' [-Winconsistent-missing-override]
    bool onTheFlyStartWalking(const double smoothingTime = 2.0);
         ^
/home/travis/build/robotology/robotology-superbuild/build/robotology/walking-controllers/modules/Walking_module/include/thrifts/WalkingCommands.h:39:16: note: 
      overridden virtual function is here
  virtual bool onTheFlyStartWalking(const double smoothingTime = 2);
               ^
In file included from /home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/src/main.cpp:14:
/home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingModule.hpp:383:18: warning: 
      'prepareRobot' overrides a member function but is not marked 'override'
      [-Winconsistent-missing-override]
    virtual bool prepareRobot(bool onTheFly = false);
                 ^
/home/travis/build/robotology/robotology-superbuild/build/robotology/walking-controllers/modules/Walking_module/include/thrifts/WalkingCommands.h:29:16: note: 
      overridden virtual function is here
  virtual bool prepareRobot(const bool onTheFly = 0);
               ^
In file included from /home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/src/main.cpp:14:
/home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingModule.hpp:389:18: warning: 
      'startWalking' overrides a member function but is not marked 'override'
      [-Winconsistent-missing-override]
    virtual bool startWalking();
                 ^
/home/travis/build/robotology/robotology-superbuild/build/robotology/walking-controllers/modules/Walking_module/include/thrifts/WalkingCommands.h:34:16: note: 
      overridden virtual function is here
  virtual bool startWalking();
               ^
In file included from /home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/src/main.cpp:14:
/home/travis/build/robotology/robotology-superbuild/robotology/walking-controllers/modules/Walking_module/include/WalkingModule.hpp:397:18: warning: 
      'setGoal' overrides a member function but is not marked 'override'
      [-Winconsistent-missing-override]
    virtual bool setGoal(double x, double y);
                 ^
/home/travis/build/robotology/robotology-superbuild/build/robotology/walking-controllers/modules/Walking_module/include/thrifts/WalkingCommands.h:45:16: note: 
      overridden virtual function is here
  virtual bool setGoal(const double x, const double y);
               ^
4 warnings and 1 error generated.
make[5]: *** [modules/Walking_module/CMakeFiles/WalkingModule.dir/src/main.cpp.o] Error 1
make[4]: *** [modules/Walking_module/CMakeFiles/WalkingModule.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [robotology/walking-controllers/CMakeFiles/YCMStamp/walking-controllers-build] Error 2
make[1]: *** [CMakeFiles/walking-controllers.dir/all] Error 2
make: *** [all] Error 2
```

traversaro, 2018-07-13:
`'auto' not allowed in function prototype`

How does GCC compile this? ?_?

S-Dafarra, 2019-01-24:
This has been fixed in the ``devel`` branch.