gbionics/walking-controllers #12: Link error occurred during installation

dosh1234, 2018-07-19:
I install it followed README.md，but I found link error

`CMakeFiles/WalkingModule.dir/src/WalkingQPInverseKinematics_qpOASES.cpp.o: In function WalkingQPIK_qpOASES::solve():
WalkingQPInverseKinematics_qpOASES.cpp:(.text+0x5130): undefined reference to 
qpOASES::SQProblem::hotstart(double const*, double const*, double const*, double const*, double const*, double const*, double const*, int&, double*, qpOASES::Bounds const*, qpOASES::Constraints const*)
WalkingQPInverseKinematics_qpOASES.cpp:(.text+0x5270): undefined reference to 
qpOASES::QProblem::init(double const*, double const*, double const*, double const*, double const*, double const*, double const*, int&, double*, double const*, double const*, qpOASES::Bounds const*, qpOASES::Constraints const*, double const*)
CMakeFiles/WalkingModule.dir/src/WalkingQPInverseKinematics_qpOASES.cpp.o: In function void __gnu_cxx::new_allocator<qpOASES::SQProblem>::construct<qpOASES::SQProblem, int&, int&>(qpOASES::SQProblem*, int&, int&):
WalkingQPInverseKinematics_qpOASES.cpp:(.text._ZN9__gnu_cxx13new_allocatorIN7qpOASES9SQProblemEE9constructIS2_JRiS5_EEEvPT_DpOT0_[_ZN9__gnu_cxx13new_allocatorIN7qpOASES9SQProblemEE9constructIS2_JRiS5_EEEvPT_DpOT0_]+0x6b): undefined reference to qpOASES::SQProblem::SQProblem(int, int, qpOASES::HessianType, qpOASES::BooleanType)
collect2: error: ld returned 1 exit status
modules/Walking_module/CMakeFiles/WalkingModule.dir/build.make:525: recipe for target 
modules/Walking_module/WalkingModule failed
make[2]: *** [modules/Walking_module/WalkingModule] Error 1
CMakeFiles/Makefile2:148: recipe for target modules/Walking_module/CMakeFiles/WalkingModule.dir/all failed
make[1]: *** [modules/Walking_module/CMakeFiles/WalkingModule.dir/all] Error 2
Makefile:127: recipe for target all failed
make: *** [all] Error 2`

I did ccmake ../ Ifound:

` CMAKE_BUILD_TYPE                                           
                                                                                                            
 CMAKE_INSTALL_PREFIX             /usr/local     
                                                                                                                       
 Eigen3_DIR                       Eigen3_DIR-NOTFOUND     
                                                                                                           
 ICUB_DIR                         /usr/local/lib/ICUB           
                                                                                                     
 ICUB_INCLUDE_DIRS                /usr/local/include;/usr/include      
                                                                                                      
 OsqpEigen_DIR                    /usr/local/lib/cmake/OsqpEigen       
                                                                                                      
 UnicyclePlanner_DIR              /usr/local/lib/cmake/UnicyclePlanner   
                                                                                                    
 YARP_DIR                         /usr/lib/x86_64-linux-gnu/cmake/YARP         
                                                                                              
 YCM_DIR                          /usr/local/share/YCM/cmake   
                                                                                                              
 iDynTree_DIR                     /usr/local/lib/cmake/iDynTree
                                                                                                              
 osqp_DIR                         /usr/local/lib/cmake/osqp    
                                                                                                              
 qpOASES_INCLUDEDIR               /usr/local/include/include  
                                                                                                               
 qpOASES_LIB                      /usr/lib/libqpOASES.so `

Why I got "undefined reference to qpOASES::**" errors, thank you!

traversaro, 2018-07-19:
Hi @dosh1234 , can you post the error that you get by running `make VERBOSE=1` ? To include the output of a given command, you can use the `~~~` to delimit the text. 

Which operating system are you using? How did you installed `qpOASES`? It is interesting that `qpOASES_LIB` points to `/usr/lib/libqpOASES.so` (i.e. <prefix> is `/usr`) while `qpOASES_INCLUDEDIR` points to `/usr/local/include/include` (i.e. <prefix> is `/usr/local`). It is possible that you have two conflicting versions of qpOASES on your machine?

dosh1234, 2018-07-19:
Thank you very much @traversaro ! I'm using Ubuntu 16.04, I installed qpOASES followed manual.pdf,(1.download qpOASES-3.2.1.zip; 2. decompression； 3. cd  install-dir && make). But then I dont know what I can do next. so I got libqpOASES.so and libqpOASES.a. I use:
sudo cp libqpOASES.a libqpOASES.so /usr/local/lib/;
cp -a ../../qpOASES-3.2.1/include/* /usr/include/.;
I don't know if it's right

Then I change ccmake value 
sudo cp libqpOASES.a libqpOASES.so /usr/lib;
cp -a ../../qpOASES-3.2.1/include/* /usr/include/.;
got                                                                 
 YCM_DIR                          /usr/local/share/YCM/cmake                                                                              
 iDynTree_DIR                     /usr/local/lib/cmake/iDynTree                                                                           
 osqp_DIR                         /usr/local/lib/cmake/osqp                                                                               
 qpOASES_INCLUDEDIR               /usr/include                                                                                            
 qpOASES_LIB                      /usr/lib/libqpOASES.so 
But still have the same link error.

I run make VERBOSE=1

/usr/bin/cmake -H/home/dosh/soft/walking-controllers -B/home/dosh/soft/walking-controllers/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/dosh/soft/walking-controllers/build/CMakeFiles /home/dosh/soft/walking-controllers/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/dosh/soft/walking-controllers/build'
make -f modules/Walking_module/CMakeFiles/icubWalking-service.dir/build.make modules/Walking_module/CMakeFiles/icubWalking-service.dir/depend
make[2]: Entering directory '/home/dosh/soft/walking-controllers/build'
cd /home/dosh/soft/walking-controllers/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/dosh/soft/walking-controllers /home/dosh/soft/walking-controllers/modules/Walking_module /home/dosh/soft/walking-controllers/build /home/dosh/soft/walking-controllers/build/modules/Walking_module /home/dosh/soft/walking-controllers/build/modules/Walking_module/CMakeFiles/icubWalking-service.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/dosh/soft/walking-controllers/build'
make -f modules/Walking_module/CMakeFiles/icubWalking-service.dir/build.make modules/Walking_module/CMakeFiles/icubWalking-service.dir/build
make[2]: Entering directory '/home/dosh/soft/walking-controllers/build'
make[2]: Nothing to be done for 'modules/Walking_module/CMakeFiles/icubWalking-service.dir/build'.
make[2]: Leaving directory '/home/dosh/soft/walking-controllers/build'

[ 11%] Built target icubWalking-service
make -f modules/Walking_module/CMakeFiles/WalkingModule.dir/build.make modules/Walking_module/CMakeFiles/WalkingModule.dir/depend
make[2]: Entering directory '/home/dosh/soft/walking-controllers/build'
cd /home/dosh/soft/walking-controllers/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/dosh/soft/walking-controllers /home/dosh/soft/walking-controllers/modules/Walking_module /home/dosh/soft/walking-controllers/build /home/dosh/soft/walking-controllers/build/modules/Walking_module /home/dosh/soft/walking-controllers/build/modules/Walking_module/CMakeFiles/WalkingModule.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/dosh/soft/walking-controllers/build'
make -f modules/Walking_module/CMakeFiles/WalkingModule.dir/build.make modules/Walking_module/CMakeFiles/WalkingModule.dir/build
make[2]: Entering directory '/home/dosh/soft/walking-controllers/build'

[ 15%] Linking CXX executable WalkingModule
cd /home/dosh/soft/walking-controllers/build/modules/Walking_module && /usr/bin/cmake -E cmake_link_script CMakeFiles/WalkingModule.dir/link.txt --verbose=1
/usr/bin/c++      CMakeFiles/WalkingModule.dir/src/main.cpp.o CMakeFiles/WalkingModule.dir/src/TrajectoryGenerator.cpp.o CMakeFiles/WalkingModule.dir/src/MPCSolver.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingController.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingDCMReactiveController.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingModule.cpp.o CMakeFiles/WalkingModule.dir/src/Utils.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingInverseKinematics.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingQPInverseKinematics_osqp.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingQPInverseKinematics_qpOASES.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingForwardKinematics.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingZMPController.cpp.o CMakeFiles/WalkingModule.dir/src/StableDCMModel.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingPIDHandler.cpp.o CMakeFiles/WalkingModule.dir/src/WalkingLogger.cpp.o CMakeFiles/WalkingModule.dir/src/TimeProfiler.cpp.o  -o WalkingModule /usr/local/lib/libctrlLib.a /usr/local/lib/libUnicyclePlanner.so /usr/local/lib/libOsqpEigen.so.0.3.0 /usr/local/lib/libosqp.so libicubWalking-service.a -lqpOASES /usr/lib/x86_64-linux-gnu/libgsl.so /usr/lib/x86_64-linux-gnu/libgslcblas.so /usr/local/lib/libidyntree-estimation.so /usr/local/lib/libidyntree-inverse-kinematics.so /usr/local/lib/libidyntree-high-level.so /usr/local/lib/libidyntree-modelio-urdf.so /usr/lib/libipopt.so /usr/lib/libdmumps_seq.so /usr/lib/libblas.so /usr/lib/liblapack.so /usr/lib/x86_64-linux-gnu/libdl.so /usr/lib/gcc/x86_64-linux-gnu/5/libgfortran.so -lm /usr/lib/gcc/x86_64-linux-gnu/5/libquadmath.so  [... cut ...]
modules/Walking_module/CMakeFiles/WalkingModule.dir/build.make:525: recipe for target 'modules/Walking_module/WalkingModule' failed
make[2]: *** [modules/Walking_module/WalkingModule] Error 1
make[2]: Leaving directory '/home/dosh/soft/walking-controllers/build'
CMakeFiles/Makefile2:148: recipe for target 'modules/Walking_module/CMakeFiles/WalkingModule.dir/all' failed
make[1]: *** [modules/Walking_module/CMakeFiles/WalkingModule.dir/all] Error 2
make[1]: Leaving directory '/home/dosh/soft/walking-controllers/build'
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

closed 2018-07-30 by dosh1234

bemilio, 2019-12-17:
I experienced the same issue to this date. @GiulioRomualdi