gbionics/walking-controllers #13: Add a warning in the README that not all iCubs are supported for walking

traversaro, 2018-07-19:
In https://github.com/robotology/walking-controllers#running-how-to-test-on-icub . 

@GiulioRomualdi @S-Dafarra

S-Dafarra, 2018-07-19:
Which limitations are you thinking about?

traversaro, 2018-07-19:
Several, in a nutshell any robot on which this is not tested has probably an aspect for which the robot will not work: 
* Robots without STRAIN2 probably experience a lot of overflow 
* Robots with legs v1 
* Configuration files are provided only for iCubGenova02 and iCubGenova04

S-Dafarra, 2018-07-19:
I see. Clearly are provided only the configuration files for the robot where we tested it, but I would say that these limitations are the same for all the controllers we developed in the lab. Regarding STRAIN2 I think we can't conclude yet that they are a bottleneck.

Anyway I will add a clarification later today. Thanks for the suggestion!

closed 2018-07-19 by S-Dafarra