gbionics/walking-controllers #15: After robotology-superbuild compiling，which models can be used for this simulation!

dosh1234, 2018-08-08:
Hi，@traversaro，I'm very sorry to disturb you！
After I installed all the dependencies, I used robotology-superbuild, turn on `ROBOTOLOGY_ENABLE_DYNAMICS`，generated install/share/gazebo under the build path, and saw a number of models in the gazebo folder. But no iCubGenova04 or iCubGenova02 or icubGazeboSim or icubGazeboV2_5 was found. Regardless of these, use `iCub (no hands)` in install/share/gazebo to continue according to `readme.md` After arriving at `yarp RPC /walking-coordinator/rpc`, An err was discovered:
`>>help`
`Responses:`
` *** Available commands:`
`  prepareRobot`
`  startWalking`
`  onTheFlyStartWalking`
`  setGoal`
`  help`
`>>prepareRobot` 
`Cannot make connection`
`yarp: Alternative method: precede port name with --client`

The `YARP_CLOCK=/clock WalkingModule` terminal display
`Segmentation fault (core dumped)`

**Is it because my robot model is wrong?** Need your help！ Thank you very much！

After run YARP_CLOCK=/clock WalkingModule, But not run  yarp RPC /walking-coordinator/rpc. I got

~$ YARP_CLOCK=/clock WalkingModule
yarp: Port /dosh-Lenovo-V310-15IKB/WalkingModule/7570/clock:i active at tcp://192.168.199.141:10088/
Success: port-to-port persistent connection added.
yarp: Waiting for clock server to start broadcasting data ...
yarp: Receiving input from /clock to /dosh-Lenovo-V310-15IKB/WalkingModule/7570/clock:i using tcp
[INFO]The model is found in:  /home/dosh/soft/robotology-superbuild/build/install/share/codyco/robots/icubGazeboSim/model.urdf 
[WARNING]  :: createReducedModelAndSensors : The joint l_leg_ft_sensor is not in the reduced model, the associated joint sensor won't be present

[WARNING]  :: createReducedModelAndSensors : The joint r_leg_ft_sensor is not in the reduced model, the associated joint sensor won't be present

[WARNING]  :: createReducedModelAndSensors : The joint l_foot_ft_sensor is not in the reduced model, the associated joint sensor won't be present

[WARNING]  :: createReducedModelAndSensors : The joint r_foot_ft_sensor is not in the reduced model, the associated joint sensor won't be present

[WARNING]  :: createReducedModelAndSensors : The joint l_arm_ft_sensor is not in the reduced model, the associated joint sensor won't be present

[WARNING]  :: createReducedModelAndSensors : The joint r_arm_ft_sensor is not in the reduced model, the associated joint sensor won't be present

[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o active at tcp://192.168.199.141:10089/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/torso/command:o active at tcp://192.168.199.141:10090/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/torso/stateExt:i active at tcp://192.168.199.141:10091/
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o to /icubSim/torso/rpc:i using tcp
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/torso/command:o to /icubSim/torso/command:i using udp
yarp: Receiving input from /icubSim/torso/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/torso/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/left_arm/rpc:o active at tcp://192.168.199.141:10092/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o active at tcp://192.168.199.141:10093/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/left_arm/stateExt:i active at tcp://192.168.199.141:10094/
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/left_arm/rpc:o to /icubSim/left_arm/rpc:i using tcp
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o to /icubSim/left_arm/command:i using udp
yarp: Receiving input from /icubSim/left_arm/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/right_arm/rpc:o active at tcp://192.168.199.141:10095/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o active at tcp://192.168.199.141:10096/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/right_arm/stateExt:i active at tcp://192.168.199.141:10097/
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/right_arm/rpc:o to /icubSim/right_arm/rpc:i using tcp
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o to /icubSim/right_arm/command:i using udp
yarp: Receiving input from /icubSim/right_arm/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/right_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/left_leg/rpc:o active at tcp://192.168.199.141:10098/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/left_leg/command:o active at tcp://192.168.199.141:10099/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/left_leg/stateExt:i active at tcp://192.168.199.141:10100/
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/left_leg/rpc:o to /icubSim/left_leg/rpc:i using tcp
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/left_leg/command:o to /icubSim/left_leg/command:i using udp
yarp: Receiving input from /icubSim/left_leg/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/left_leg/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/right_leg/rpc:o active at tcp://192.168.199.141:10101/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/right_leg/command:o active at tcp://192.168.199.141:10102/
yarp: Port /walking-coordinator/remoteControlBoard/icubSim/right_leg/stateExt:i active at tcp://192.168.199.141:10103/
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/right_leg/rpc:o to /icubSim/right_leg/rpc:i using tcp
yarp: Sending output from /walking-coordinator/remoteControlBoard/icubSim/right_leg/command:o to /icubSim/right_leg/command:i using udp
yarp: Receiving input from /icubSim/right_leg/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/right_leg/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]created device <remotecontrolboardremapper>. See C++ class yarp::dev::RemoteControlBoardRemapper for documentation.
yarp: Port /walking-coordinator/leftFootWrench:i active at tcp://192.168.199.141:10104/
yarp: Receiving input from /wholeBodyDynamics/left_foot/cartesianEndEffectorWrench:o to /walking-coordinator/leftFootWrench:i using tcp
yarp: Port /walking-coordinator/rightFootWrench:i active at tcp://192.168.199.141:10105/
yarp: Receiving input from /wholeBodyDynamics/right_foot/cartesianEndEffectorWrench:o to /walking-coordinator/rightFootWrench:i using tcp
yarp: Port /walking-coordinator/rpc active at tcp://192.168.199.141:10106/
[INFO]Loading custom PIDs
[INFO]Parsing DEFAULT PID group. 
[INFO]DEFAULT PID successfully loaded
[INFO]Option 	 Value 
[INFO]pos filt 	  false 
[INFO]vel filt 	  false 
[INFO]wre filt 	  false 
[INFO]useMPC 	  true 
[INFO]useQPIK 	  true 
[INFO]useOSQP 	  false 
[INFO]dump 	  false 
[INFO]velocity 	  false 
[INFO][configure] Ready to play!

traversaro, 2018-08-08:
Hi @dosh1234 , 

 how did you set the `YARP_ROBOT_NAME` env variable?

dosh1234, 2018-08-08:
@traversaro ,Yes, I set YARP_ROBOT_NAME="icubGazeboSim". I write `export YARP_ROBOT_NAME="icubGazeboSim"` in  ~/..bashrc

traversaro, 2018-08-08:
Then I guess the configuration is ok. To gather more information, you can compile the `walking-controllers` repo in debug, and you can run `WalkingModule` under `valgrind` to get more information about the actual part of the code that is creating the segfault. 

@S-Dafarra @GiulioRomualdi any idea?

GiulioRomualdi, 2018-08-08:
Hi @dosh1234, the model seems fine. I think that the problem is related to `qpOASES`. 
I try to explain what may happen. When you send `prepareRobot` through the RPC port a non-linear optimization problem is solved by using `IPOPT`. `IPOPT` should call `dpotrf_` function implemented in `LAPACK`. However, when `qpOASES` and `IPOPT` are linked in the same binary `IPOPT` calls the `dpotrf_` function implement in `qpOASES` instead of the correct one in `LAPACK` as consequence the application crashes.
You can find further information here https://github.com/robotology-dependencies/qpOASES/pull/7

In order to understand if this is the cause of your problem can you check if the `qpOASES` library found by 'cmake` is the one downloaded by the `robotology-superbuild`?

The following variable should be set as follows:
```
qpOASES_INCLUDEDIR        path/where/you/installed/robotology-superbuild/build/install/include 
qpOASES_LIB               path/where/you/installed/robotology-superbuild/build/install/lib/libqpOASES.so
```

dosh1234, 2018-08-08:
Hi @GiulioRomualdi， 
Which version of qpOASES is required? I found that walking-controllers uses the 3.2 version of qpOASES will meet errors when Compiling. The error is as follows：

WalkingQPInverseKinematics_qpOASES.cpp:(.text+0x5164): undefined reference to qpOASES::SQProblem::hotstart(double const*, double const*, double const*, double const*, double const*, double const*, double const*, int&, double*, qpOASES::Bounds const*, qpOASES::Constraints const*)

I find that the qpOASES::SQProblem::hotstart function parameter in  3.2 version of qpOASES is different from it in 3.0. if I used 3.0 compilation will be successful. 

Because I didn't turn on  `ROBOTOLOGY_USES_MATLAB`, so the robotology-superbuild didn't downloaded qpOASES, I downloaded  qpOASES by 
`svn co https://projects.coin-or.org/svn/qpOASES/stable/3.x qpOASES`
`cd <install-dir>`
`make`
and
`sudo cp -r ~/soft/qpOASES_3.x/include/* /usr/include/`
`sudo cp ~/soft/qpOASES_3.x/bin/libqpOASES.* /usr/lib`
then
`cd /walking-controllers/build && ccmake ../`
set
`qpOASES_INCLUDEDIR      /usr/include    `                                                                                                                                              
`qpOASES_LIB                     /usr/lib/libqpOASES.so`

traversaro, 2018-08-08:
Unfortunately only the trunk version of qpOASES will work, as it is the only one actually containing the fix https://projects.coin-or.org/qpOASES/changeset/255/ (see https://github.com/robotology-dependencies/qpOASES/issues/8 for more on this). Alternatively, you can use our fork at https://github.com/robotology-dependencies/qpOASES that should work fine (it should be based on 3.2, so I was surprised that you had API problems and I believe there is some mismatch between the headers and the library).

dosh1234, 2018-08-10:
Hi @traversaro ,walking-controllers's link error has been solved, I open the ROBOTOLOGY_USES_MATLAB to use superbuild, and then connect to the 
`robotology-superbuild/build/install/lib/libqpOASES.so.`

But there's another problem. Follow the readme.md and run:
`YARP_CLOCK=/clock WalkingModule`

Got errs:
......
yarp: Port /walking-coordinator/leftFootWrench:i active at tcp://192.168.199.141:10101/
yarp: Receiving input from /wholeBodyDynamics/left_foot/cartesianEndEffectorWrench:o to /walking-coordinator/leftFootWrench:i using tcp
yarp: Port /walking-coordinator/rightFootWrench:i active at tcp://192.168.199.141:10102/
yarp: Receiving input from /wholeBodyDynamics/right_foot/cartesianEndEffectorWrench:o to /walking-coordinator/rightFootWrench:i using tcp
yarp: Port /walking-coordinator/rpc active at tcp://192.168.199.141:10103/
[ERROR][getNumberFromSearchable] Missing field  kIDCM 
[ERROR][initialize] Unable to get the double from searchable. 
[ERROR][configure] Unable to initialize the controller. 
[INFO]RFModule failed to open.
yarp: Removing input from /wholeBodyDynamics/right_foot/cartesianEndEffectorWrench:o to /walking-coordinator/rightFootWrench:i
yarp: Removing input from /wholeBodyDynamics/left_foot/cartesianEndEffectorWrench:o to /walking-coordinator/leftFootWrench:i
yarp: Removing output from /walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o to /icubSim/torso/rpc:i
......

What's wrong with me again?

GiulioRomualdi, 2018-08-10:
Hi! Which branch are you using?

dosh1234, 2018-08-10:
@GiulioRomualdi I used `walking-controllers-Humanoids2018`,and robotology-superbuild used `master`.
Thank you very much!

GiulioRomualdi, 2018-08-10:
Ok, probably it is my fault. Let me check

dosh1234, 2018-08-10:
@GiulioRomualdi If I used `master`, everything will be fine. Thank you very much! @traversaro and @GiulioRomualdi . What do you recommend for information about this project, such as software architecture or tutorials.Thank you again.

GiulioRomualdi, 2018-08-10:
@dosh1234, now also the `Humanoids2018` branch should work.

GiulioRomualdi, 2018-09-26:
@dosh1234 have you tried again?

dosh1234, 2018-09-27:
@GiulioRomualdi ，Yes，I have tried again. `Humanoids2018` has already been used. I regret not having informed you in time.
Recently, I have been doing research on inverse kinematics of robot based on CoM trajectory and swing leg trajectory. I don't know what algorithm the `walking controller` adopted on this issue.Looking forward to the theoretical knowledge behind the `walking controller`.

GiulioRomualdi, 2018-09-27:
Hi! Here https://arxiv.org/abs/1809.02167 , you can find the theory behind the architecture

GiulioRomualdi, 2018-09-27:
I'll close the issue. Do not esitate to open it again

closed 2018-09-27 by GiulioRomualdi