gbionics/walking-controllers #19: .urdf models are not installed

S-Dafarra, 2018-10-12:
From the CMakeLists (https://github.com/robotology/walking-controllers/blob/devel_hand_retargeting/app/CMakeLists.txt#L22-L28) it seems that models are not installed.

traversaro, 2018-10-12:
Can we track in  https://github.com/robotology/icub-model-generator (or in an issue here) why do you need custom models? Software-specific model create a maintenance nightmare whenever we need to update the robot model, as any upgrade (such as a different weight in a battery or a different ankle) needs to be manually ported to every repo.

GiulioRomualdi, 2018-10-12:
We added this model because the one found by the resource finder was wrong. If I remember correctly, some months ago the model found was not the one in icub-models but the one in codyco-modules.

traversaro, 2018-10-12:
`icub-models` is now included in the superbuild, and t he models in `yarp-wholebodyinterface` were removed in https://github.com/robotology/yarp-wholebodyinterface/pull/90 . Can you check if this is still necessary, and if it is not remove the `urdf` from the repo? Thanks.

GiulioRomualdi, 2018-10-12:
Yes!

commit db596584 (2018-10-25): merge branch `devel` into `master`
This:
* implements ZMP/CoM gain scheduling
* Fixes #19
* allows using `make uninstall`
* moves `Utilities` in a separate library
commit 1f4f67dc (2018-10-25): merge branch `devel` into `master`
This:
* implements ZMP/CoM gain scheduling
* Fixes #19
* allows using `make uninstall`
* moves `Utilities` in a separate library
* bugfix in StableDCMModel.cpp
closed 2018-10-25 by GiulioRomualdi