gbionics/walking-controllers #22: Map the joypad buttons to rpc commands

GiulioRomualdi, 2018-12-07:
To simplify the setup of the demo we should map the joypad buttons to the rpc port (e.g. press A to prepare the robot, press B to start walking ...)

DanielePucci, 2018-12-07:
This is really useful. Probably @kouroshD can take upon this?

kouroshD, 2018-12-07:
@DanielePucci  Yes, it is a good idea. when I get acquainted with the software, we can surely do it.
@GiulioRomualdi  Just the matter of curiosity, what are the functionalities of joypad buttons now?

kouroshD, 2018-12-10:
@DanielePucci yes, it is a good idea, when I start implementing, I will do this as well.

GiulioRomualdi, 2018-12-10:
@kuroshD, right now the buttons have no functionality. The main utility of the joypad is sending the desired walking orientation and speed by means of the left joipad axis. 

To understand if one of the button is pressed you may use [’getButton’](http://www.yarp.it/classyarp_1_1dev_1_1IJoypadController.html#a71875d6605e7deb30003587f898b8321) method.

GiulioRomualdi, 2018-12-27:
Done in https://github.com/robotology/walking-controllers/commit/df831d855e834b67309547a87bb368285287b2d9

commit df831d85 (2018-12-18): - add the possibility of using directly the SDLJoypad device instead of the yarp JoypadControlClient
- close #22 map the buttons to rpc commands ("A" for preparing the robot, "B" for strat walking and "L1 + R1" for connecting the JoypadModule to the the WalkingModule)
PR #27 (2019-01-25): Merge devel branch in master 
This PR
- Implements the base class for `QpIK`
- adds the possibility of using directly the `SDLJoypad` device instead of the `JoypadControlClient`
- close #22 map the buttons to rpc commands ("A" for preparing the robot, "B" for strat walking and "L1 + R1" for connecting the JoypadModule to the the WalkingModule ... )
- allows to use [last verson](https://github.com/robotology/unicycle-footstep-planner/releases/tag/v0.1.0) of the uncycicle planner
- remove usage of `YARP_INCLUDE_DIRS`
- close #20 
- update the `README.md`
closed 2019-01-25 by GiulioRomualdi