gbionics/walking-controllers #34: Move WalkingLogger and Utils to a common location

prashanthr05, 2019-06-03:
The `WalkingLogger` module and `Utils` library which was originally by @GiulioRomualdi created as a part of the `walking-controllers` repository. Eventually, these were duplicated also for the floating base estimation and is meant to be used as a part of `whole-body-estimators` repository as well. 

In reference to this comment https://github.com/robotology/whole-body-estimators/pull/1#issuecomment-486958679, in order to avoid duplication of code, it would be better to have these utilities in a common location accessible by both `walking-controllers` and `whole-body-estimators`. 

Also, the usefulness of these utilities extend beyond these two repositories and can be used by other users as well.

GiulioRomualdi, 2019-06-03:
For the `WalkingLogger` I suggest using https://github.com/ADVRHumanoids/MatLogger2. I think is more versatile than the `WalkingLogger` 

cc @traversaro

traversaro, 2019-06-03:
I really like MatLogger2 as well (the license still needs to be documented, but I think it will be quite a liberal one). For the `Utils` class, I briefly checked and all the iDynTree helpers one I think they can be moved in iDynTree itself.

GiulioRomualdi, 2019-06-03:
As regards the yarp helpers? Should we move them in a separate repo?

traversaro, 2019-06-03:
It depends. 

I guess that:  `yarpListToiDynTreeVectorFixSize` and any other iDynTree <--> YARP conversion function can go in iDynTree's YARP component, i.e. in https://github.com/robotology/idyntree/blob/master/src/yarp/include/iDynTree/yarp/YARPConversions.h and https://github.com/robotology/idyntree/blob/master/src/yarp/include/iDynTree/yarp/YARPConfigurationsLoader.h .

For most of the other YARP-only helpers, I think we should  be able to put them in YARP, but we would need to check with @drdanz for guessing the right location were to place them.

GiulioRomualdi, 2019-06-11:
See #35

drdanz, 2019-06-13:
> For most of the other YARP-only helpers, I think we should be able to put them in YARP, but we would need to check with @drdanz for guessing the right location were to place them.

Sorry, I missed this... Are [these](https://github.com/robotology/walking-controllers/blob/master/modules/Utilities/include/Utils.hpp) the utilities you are talking about? I'd say they could be somewhere in yarp::sig

prashanthr05, 2019-06-24:
@drdanz There are `yarp::os` related functions like 
- `bool addVectorOfStringToProperty(yarp::os::Property& prop, 
                                                           const std::string& key,
                                                           const std::vector<std::string>& list)`, 
- `bool yarpListToStringVector(yarp::os::Value*& input, std::vector<std::string>& output)`
- `bool getStringFromSearchable(const yarp::os::Searchable& config, const std::string& key,
                                 std::string& string)`
- `bool getDoubleFromSearchable(const yarp::os::Searchable& config, const std::string& key,
                                 double& number)`
- `bool getIntegerFromSearchable(const yarp::os::Searchable& config, const std::string& key,
                                  int& number)`
- `void populateBottleWithStrings(yarp::os::Bottle& bottle, const std::initializer_list<std::string>& strings);`

and there are `yarp::sig` related functions like,
-  `template <typename T>
void mergeSigVector(yarp::sig::Vector& vector, const T& t);`
- `template <typename T, typename... Args>
void mergeSigVector(yarp::sig::Vector& vector, const T& t, const Args&... args);`
- `template <typename... Args>
    void sendVariadicVector(yarp::os::BufferedPort<yarp::sig::Vector>& port, const Args&... args);`

Would it be meaningful to open separate PRs for `sig` and `os` ?

GiulioRomualdi, 2022-09-22:
All these features have been moved in bipedal-locomotion-framework [here](https://github.com/ami-iit/bipedal-locomotion-framework/tree/master/src/YarpUtilities)

closed 2022-09-22 by GiulioRomualdi