gbionics/walking-controllers #35: Embed MatLogger2 into walking-controllers

GiulioRomualdi, 2019-06-11:
It would be nice to use [MatLogger2](https://github.com/ADVRHumanoids/MatLogger2) inside walking-controllers. This library can substitute the WalkingLogger library.

cc @traversaro

traversaro, 2019-06-12:
To avoid symbol clashes (as described in https://github.com/robotology/how-to-export-cpp-library/issues/46#issuecomment-500135506) something that we should make sure is that MatLogger2 should be able to consume a system matio library, instead of using its own vendored version. We can open an issue upstream for that.

traversaro, 2019-06-12:
Another thing to check is macOS and Windows compatibility.

GiulioRomualdi, 2022-09-22:
This is not required anymore after https://github.com/robotology/walking-controllers/pulls?q=is%3Apr+is%3Aclosed+vectorscollection

closed 2022-09-22 by GiulioRomualdi