gbionics/walking-controllers #36: Error for building walking controller module because of osqpEigen v 0.4.0

kouroshD, 2019-06-11:
The `devel` branch of walking controller gives this error:
```
CMake Error at modules/Walking_module/CMakeLists.txt:20 (find_package):
  Could not find a configuration file for package "OsqpEigen" that is
  compatible with requested version "0.4.0".

  The following configuration files were considered but not accepted:

    /Users/kdarvish/Desktop/icub_ws/robotology-superbuild/build/install/lib/cmake/OsqpEigen/OsqpEigenConfig.cmake, version: 0.3.0
```

GiulioRomualdi, 2019-06-12:
Hi @kouroshD, `walking-controller` in `devel` needs osqp-eigen v0.4.0, i.e. osqp-eigen has to be in devel.

osqp-eigen [master](https://github.com/robotology/osqp-eigen/blob/master/CMakeLists.txt#L28-L30)
```cmake
project(OsqpEigen
  LANGUAGES CXX
  VERSION 0.3.0)
```

osqp-eigen [devel](https://github.com/robotology/osqp-eigen/blob/devel/CMakeLists.txt#L28-L30)
```cmake
project(OsqpEigen
  LANGUAGES CXX
  VERSION 0.4.0)
```

traversaro, 2019-06-12:
@GiulioRomualdi could it make sense to add this two `devel` branches in https://github.com/robotology/robotology-superbuild/blob/master/cmake/ProjectsTagsDevel.cmake ?

closed 2022-09-22 by GiulioRomualdi