gbionics/walking-controllers #37: Enhance the Walking Controller Logger

kouroshD, 2019-07-12:
Currently, walking logger logs mainly the locomotion data, while to debug the problems,  we may need other information of the controller such as:
- all joint angles & velocities
- end effector pose & velocities

cc @fjandrad @GiulioRomualdi  @DanielePucci

S-Dafarra, 2019-07-12:
I am getting convinced that having a powerful logger is very important if we want to increase the overall performances of the algorithms we are running on the robot. Maybe we can take this a step further and start developing a standalone application able to log all the relevant data, like internal states of the controllers, commands sent and feedbacks.

kouroshD, 2019-07-12:
I completely agree with that.

kouroshD, 2019-07-12:
To add what you said, it can be useful also adding a Matlab script, with options to enable, for visualizing the data. Because, in this case it is not recommended logging all the data in one single file, since it can become easily large size.

S-Dafarra, 2019-07-12:
I don't think the file size is an issue. Right now hard disks are pretty cheap, and we could use a dedicated machine for logging. For the data visualization, a link that appeared a couple of times in some previous discussions is https://github.com/facontidavide/PlotJuggler (I never tried it). I think the possibility to drag and drop the data is pretty useful. For logging data, we could exploit https://github.com/ADVRHumanoids/MatLogger2

GiulioRomualdi, 2019-07-12:
We should develop something like `coredumpctl`

S-Dafarra, 2019-07-12:
What do you mean? :thinking:

GiulioRomualdi, 2019-07-12:
Suppose that you are running something in the terminal. At the certain point a segmentation fault occurs. Using `coredumpctl gdb` you can understand the reason of the problem without running again the application.

S-Dafarra, 2019-07-12:
Interesting, but how do you think to use it in practice to log data?

GiulioRomualdi, 2019-07-12:
Ehm  I don't know

fjandrad, 2019-07-16:
> Suppose that you are running something in the terminal. At the certain point a segmentation fault occurs. Using coredumpctl gdb you can understand the reason of the problem without running again the application.

With @prashanthr05 we are working to get an offline environment in which you can replay data logged with data dumper and run your application on that data as if the robot was working. Not sure if it makes sense, but we could start from something like and build up to something more powerful for logging/debugging and testing.

GiulioRomualdi, 2019-09-11:
This morning I started to work on a new version of the Logger. The logger is based on [`MatLogger2`](https://github.com/ADVRHumanoids/MatLogger2).

You can find the code here: https://github.com/GiulioRomualdi/YarpMatLogger

GiulioRomualdi, 2019-09-23:
In the last days, I worked on the logger:  https://github.com/GiulioRomualdi/YarpMatLogger. 
Since the logger does not depend upon the walking and it can be used also by different modules, probably we should finf a proper repo where store the code.

fjandrad, 2019-10-21:
I would consider the logger enhanced. Also the title is a bit generic suited for an epic xD. We might consider closing this issue and then open specific issues for where to put the logger and individual features we want to add to it.

S-Dafarra, 2019-10-29:
@GiulioRomualdi please decide if you want to substitute the current logger but let's not put too much effort on this. Then close this issue.

GiulioRomualdi, 2019-12-03:
I will move the new logger in the new walking-repo https://github.com/dic-iit/bipedal-locomotion-controllers

GiulioRomualdi, 2022-09-22:
Closing after https://github.com/robotology/walking-controllers/pull/97

closed 2022-09-22 by GiulioRomualdi