gbionics/walking-controllers #42: Remove the dependecy of yarp from the walking-controllers libraries

GiulioRomualdi, 2019-10-01:
I think that one of the main drawbacks of the current implementation of the walking-controller is how the blocks are connected together. 
In details:
1. there are no libraries: The main components of the walking should become libraries and the `WalkingModule` should be an application that loads only `shared` or `static` libraries.
2. the submodules of the walking (i.e. the inverse kinematics) should not depend upon `yarp`. This independence could be useful if someone wants to use our libraries in a `yarp-free` architecture. Currently, the main dependence of `yarp` are: the `resource finder` and the  [`minJerkTraj`](http://www.icub.org/software_documentation/group__minJerkCtrl.html). 
For the former, we could use a _standard_ library for loading parameters from configuration files — for instance [`libconfig`](https://github.com/hyperrealm/libconfig) which is compatible with Linux MacOs and Windows. For the latter, we could port the `minJerkTraj` into `iDynTree`. 

This approach could be one of the first steps through the #30, it should also allow using the controller not only on the `GazeboSim` but also on other simulators (i.e. [`Choreonoid`](https://choreonoid.org/en/)).

What do you think @S-Dafarra @traversaro @MiladShafiee @prashanthr05? Do you have any ideas?

S-Dafarra, 2019-10-01:
Regarding the parameters, we could think about a separate class containing all the fields you need inside the controller. Then you can fill this struct in a different way depending on the application. In this way, the single components don't depend neither on ``yarp`` nor on ``libconfig``, but the application that merges them all will.

traversaro, 2019-10-01:
Regarding parameters, a possible alternative approach (that could be made compatible with what proposed by @S-Dafarra) is to **declare** which parameters a block has, and just have a single function/class to load this parameters from a YARP .ini group/json file/yaml file or whatever format you prefer. 
Example of this pattern:
* https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_parameters.html 
* https://robotology.github.io/blockfactory/doxygen/classblockfactory_1_1core_1_1_parameter_metadata.html 
*  https://fmi-standard.org/docs/2.0.1-develop/#_getting_and_setting_variable_values

traversaro, 2019-10-01:
> the submodules of the walking (i.e. the inverse kinematics) should not depend upon yarp. This independence could be useful if someone wants to use our libraries in a yarp-free architecture. Currently, the main dependence of yarp are: the resource finder and the minJerkTraj.
For the former, we could use a standard library for loading parameters from configuration files — for instance libconfig which is compatible with Linux MacOs and Windows. For the latter, we could port the minJerkTraj into iDynTree.

Note that I would not mix the concern of "refactor walking-controllers as libraries" and "remove yarp dependencies". These are two important and useful improvements, but it is important to make clear that they do not depend too much on each other. Even if we move to something non-YARP for dealing with parameters (the biggest part that depend on YARP), I think it would make sense to still have ã few ~~blocks~~ submodules that depend on YARP, similarly to the situation in WB-Toolbox. Then the compilation for those ~~blocks~~ submodules can be disabled if necessary.

prashanthr05, 2019-10-01:
Usually, people with ROS background develop their algorithms as stand-alone classes and then additionally create a ROS wrapper in order to do the configuration step.

Getting inspired also from the BlockFactory workflow and also ROS wrappers, I refactored the base estimators like this, 
https://github.com/dic-iit/element_floating-base-estimation/issues/85#issue-493520220
[image]

Where, the Estimator API is basically the algorithms in a BlockFactory view. (configure, set input, get output and dynamic reconfigure (idea from ROS)).

The central block is the YARP wrapper, where I run the YARP device using a Periodic thread, that instantiates objects of classes `RobotInterface` and `BaseEstimatorInterface`. `RobotInterface` is a YARP bridge to get the measurements from the robot and `BaseEstimatorInterface` is basically the parameter bridge used to configure the Estimator algorithm from the Estimators API.

Maybe this could be of some help

GiulioRomualdi, 2019-10-01:
> Note that I would not mix the concern of "refactor walking-controllers as libraries" and "remove yarp dependencies". These are two important and useful improvements, but it is important to make [...]

Ideally, some blocks (e.g. RobotHelper) will depend on yarp. Other such that `WalkingQPIK` will be yarp-free

GiulioRomualdi, 2019-10-01:
Probably, it is easier to have a F2F meeting for defining the desiderata @traversaro @S-Dafarra @prashanthr05

traversaro, 2019-10-01:
> Ideally, some blocks (e.g. RobotHelper) will depend on yarp. Other such that WalkingQPIK will be yarp-free

I totally agree, I just wanted to make sure that we properly separate the problems.

traversaro, 2019-10-03:
> Probably, it is easier to have a F2F meeting for defining the desiderata @traversaro @S-Dafarra @prashanthr05

Great idea. As my availability is quite variable, feel free to do that without me. I would be happy to provide feedback directly on more issues and to the code.

GiulioRomualdi, 2020-12-01:
Since we moved our effort into [`bipedal-locomotion-framework`](https://github.com/dic-iit/bipedal-locomotion-framework) we can close this issue.

closed 2020-12-01 by GiulioRomualdi