gbionics/walking-controllers #47: Unable to compile the repository on Windows

GiulioRomualdi, 2020-01-24:
When I try to compile the repository on windows, I retrieve the following error

```
C:\Users\gromualdi\robot-code\robotology-superbuild\robotology\walking-controllers\modules\Walking_module\include\WalkingQPInverseKinematics.hpp(34,1): warning C4267: 'argument': conversion from 'size_t' to 'const unsigned int', possible loss of data
5>C:\Program Files\robotology\eigen-3.3.4\Eigen\src/SparseCore/SparseDiagonalProduct.h(75,1): error C2248: 'Eigen::internal::unary_evaluator<T,Eigen::internal::IteratorBased,double>::InnerIterator::Scalar': cannot access private typedef declared in class 'Eigen::internal::unary_evaluator<T,Eigen::internal::IteratorBased,double>::InnerIterator'
5>        with
5>        [
5>            T=Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>,const Eigen::Map<Eigen::SparseMatrix<double,0,int>,0,Eigen::Stride<0,0>>>
5>        ]
```

I'm currently using Eigen 3.3.4 installed from [here](http://www.yarp.it/download.html#download_windows)

@traversaro  any suggestion?

traversaro, 2020-01-24:
> @traversaro any suggestion?

Unfortunately I am not certain about this due to https://github.com/robotology/robotology-superbuild/issues/226 that is preventing the GitHub Actions CI to run on Windows, but recently I think compiled walking-controllers on Windows with Visual  Studio 2019 using Eigen 3.3.7 . You can install via vcpkg or via the pre-built vcpkg workspace available at https://github.com/robotology-playground/robotology-superbuild-dependencies/releases/tag/v0.0.2, that is what we currently use for CI in robotology-superbuild (see https://github.com/robotology/robotology-superbuild/blob/master/.github/workflows/ci.yml#L86) and we would like to document as the canonical way to install deps, to remove the super-error-prone existing documentation on how to install deps.

traversaro, 2020-02-01:
I experienced the same error in GitHub Actions, see: 
~~~
2020-02-01T15:57:44.3838038Z D:\a\robotology-superbuild\robotology-superbuild\robotology\walking-controllers\modules\Walking_module\include\WalkingQPInverseKinematics.hpp(33,26): warning C4018: '<': signed/unsigned mismatch [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\walking-controllers\modules\Walking_module\WalkingModule.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\walking-controllers.vcxproj]
2020-02-01T15:57:45.2084855Z C:\vcpkg-robotology\installed\x64-windows\include\eigen3\Eigen\src/SparseCore/SparseDiagonalProduct.h(75,1): error C2248: 'Eigen::internal::unary_evaluator<T,Eigen::internal::IteratorBased,double>::InnerIterator::Scalar': cannot access private typedef declared in class 'Eigen::internal::unary_evaluator<T,Eigen::internal::IteratorBased,double>::InnerIterator' [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\walking-controllers\modules\Walking_module\WalkingModule.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\walking-controllers.vcxproj]
~~~

traversaro, 2020-02-01:
I guess this was related: http://manao.inria.fr/eigen_tmp/pullrequests/696/bb_dump.html .

traversaro, 2020-02-17:
I have the same error against the latest Eigen, so it is not a problem of a bug that was not fixed.

traversaro, 2020-02-17:
Actually the problem seems to be fixed by backporting https://gitlab.com/libeigen/eigen/-/commit/f59bed7a133322955dac03f3654def706aff3ba6 to 3.3.4, so it is possible that the latest  Eigen has another problem. However, it seems to be an issue on Eigen or in MSVC. We should probably open an issue in Eigen for that, but I can prepare a workaround for Eigen 3.3.

traversaro, 2020-02-18:
Upstream pull request for Eigen 3.3 branch: https://gitlab.com/libeigen/eigen/-/merge_requests/63 .

traversaro, 2020-03-16:
I think this was fixed by https://github.com/robotology/walking-controllers/pull/53 and  https://github.com/robotology/walking-controllers/pull/54, and upstream (pending a Eigen 3.3 patch release) in https://gitlab.com/libeigen/eigen/-/merge_requests/63 .

GiulioRomualdi, 2020-04-09:
Accordingly to https://github.com/robotology/walking-controllers/issues/47#issuecomment-599425291 we can close the issue

closed 2020-04-09 by GiulioRomualdi