gbionics/walking-controllers #50: Use the desired values of joints angles in inverse kinematic

MiladShafiee, 2020-02-06:
Here we will discuss whether should we use the desired values of joint angles in inverse kinematic function([this line of code](https://github.com/robotology/walking-controllers/blob/daaf28a15b0e8b10dca50a95cecd7c2bc6e5f15e/src/WalkingModule/src/Module.cpp#L385))?

Please notice that in [this line of code](https://github.com/robotology/walking-controllers/blob/daaf28a15b0e8b10dca50a95cecd7c2bc6e5f15e/src/WalkingModule/src/Module.cpp#L385) we use the real value for setting the current state for inverse kinematic but[ in this part of code](https://github.com/robotology/walking-controllers/blob/daaf28a15b0e8b10dca50a95cecd7c2bc6e5f15e/src/WalkingModule/src/Module.cpp#L720) we try to use the desired joints values for the inverse kinematic and it seems they are not consistent.

S-Dafarra, 2020-02-06:
It's true, there is a discrepancy. All the quantities obtained through the ``FKSolver`` are computed using the desired values, hence [these other inputs](https://github.com/robotology/walking-controllers/blob/daaf28a15b0e8b10dca50a95cecd7c2bc6e5f15e/src/WalkingModule/src/Module.cpp#L386-L391) are obtained through the desired joint values. 

I would suggest adding a method to the [``WalkingFK``](https://github.com/robotology/walking-controllers/blob/daaf28a15b/src/KinDynWrapper/include/WalkingControllers/KinDynWrapper/Wrapper.h#L27) class called ``getJointPosition`` such that [these lines of code](https://github.com/robotology/walking-controllers/blob/daaf28a15b0e8b10dca50a95cecd7c2bc6e5f15e/src/WalkingModule/src/Module.cpp#L385-L391) would become 
```c++
    ok &= solver->setRobotState(m_FKSolver->getJointPosition(),
                                m_FKSolver->getLeftFootToWorldTransform(),
                                m_FKSolver->getRightFootToWorldTransform(),
                                m_FKSolver->getLeftHandToWorldTransform(),
                                m_FKSolver->getRightHandToWorldTransform(),
                                m_FKSolver->getNeckOrientation(),
                                m_FKSolver->getCoMPosition());
```
thus avoiding the discrepancy. What do you think @MiladShafiee @GiulioRomualdi?

MiladShafiee, 2020-02-06:
> I would suggest adding a method to the WalkingFK class called getJointPosition such that these lines of code would become

I think it would be fine. Just for understanding that `m_FKSolver->getJointPosition()` are real or desired values, we probably should know that which joint values(real or desired) has been set to the `m_FKSolver` last time.

S-Dafarra, 2020-02-06:
That would automatically depend on 
https://github.com/robotology/walking-controllers/blob/daaf28a15b0e8b10dca50a95cecd7c2bc6e5f15e/src/WalkingModule/src/Module.cpp#L720

MiladShafiee, 2020-02-06:
> That would automatically depend on

Yes exactly but here handled with this part because we already knew that. If we want to use it in some other part of code we should check that which values has been set last time. 

What do you think about adding a Boolean parameter (or something like this) in `m_FKSolver` that will show us that currently is it using the real value or desired?

S-Dafarra, 2020-02-06:
> What do you think about adding a Boolean parameter (or something like this) in m_FKSolver that will show us that currently is it using the real value or desired?

I don't like it too much as there may be other versions of the joint values. In some cases, we may need to fed the ``FKSolver`` with the smoothed version of the desired joint values, in other cases the measured values where (for some reason) we modified manually the value of a joint.

In other words, the boolean would not be sufficient to understand which values are actually used.

I believe that in this case, the easiest and cleanest way to deal with this problem would be to add a comment in the code reminding which values are used for which purpose. I.e. 
```c++
 if(!m_FKSolver->setInternalRobotState(m_qDesired, m_dqDesired)) //The QP IK uses the desired values as feedback
```

GiulioRomualdi, 2020-02-06:
The fk solver should not provide the joint values. I don't understand why you want to add that function.

GiulioRomualdi, 2020-02-06:
`solver->setRobotState(m_qDesired ... )` should be fine. No?

S-Dafarra, 2020-02-06:
> I don't understand why you want to add that function.

To have the result of [the comment above](https://github.com/robotology/walking-controllers/issues/50#issuecomment-582975263). In this way, we are sure that all the data used inside the IK comes from the same source.

> ``solver->setRobotState(m_qDesired ... ) should be fine. No?``

Then, how can you be sure that the other quantities you get from the ``FKsolver`` are computed using the desired quantities? The same problem could re-appear.

S-Dafarra, 2020-02-06:
> It is only a wrapper around kindyn

Actually, ``KinDynComputations`` has [``getJointPos``](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1KinDynComputations.html#a3154c238fcccc126e56d95e40e2fd832)

GiulioRomualdi, 2020-05-13:
In https://github.com/robotology/walking-controllers/pull/41 I changed the signature of  `WalkingQPIK::setRobotState`. Now the only input is a `kinDynWrapper` object and in this case, we are sure that all the quantities used inside the `QP-IK` are computed using only desired (or if the user wants measured) quantities. 
https://github.com/robotology/walking-controllers/blob/2d0a4a5e0c851b05c001b15b17abadf2aea257fe/src/WholeBodyControllers/src/QPInverseKinematics.cpp#L388-L401

MiladShafiee, 2020-05-20:
We close this issue  because it has been solved in #41.

closed 2020-05-20 by MiladShafiee