gbionics/walking-controllers #58: Unable to compile this repository on Windows strikes back

traversaro, 2020-03-16:
Recent scheduled CI jobs of the superbuild started failing due on Windows, see https://github.com/robotology/robotology-superbuild/actions/runs/56405001 . This is probably due to https://github.com/robotology/walking-controllers/pull/55, as at the moment the superbuild uses the master branch of walking-controllers both for stable and unstable branches. 

The error in particular seems to be:
~~~
2020-03-16T02:39:46.9396578Z LINK : fatal error LNK1104: cannot open file '..\YarpUtilities\Debug\WalkingControllersYarpUtilities.lib' [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\walking-controllers\src\LoggerClient\LoggerClient.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\walking-controllers.vcxproj]
2020-03-16T02:39:47.0120257Z     Building Custom Rule D:/a/robotology-superbuild/robotology-superbuild/robotology/walking-controllers/src/RetargetingHelper/CMakeLists.txt
2020-03-16T02:39:47.0925924Z     Helper.cpp
2020-03-16T02:39:48.1668559Z D:\a\robotology-superbuild\robotology-superbuild\build\install\include\iDynTree/Core/EigenHelpers.h(275,14): warning C4244: 'argument': conversion from 'const ptrdiff_t' to 'const unsigned int', possible loss of data [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\walking-controllers\src\RetargetingHelper\RetargetingHelper.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\walking-controllers.vcxproj]
2020-03-16T02:39:48.7331865Z LINK : fatal error LNK1104: cannot open file '..\YarpUtilities\Debug\WalkingControllersYarpUtilities.lib' [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\walking-controllers\src\RetargetingHelper\RetargetingHelper.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\walking-controllers.vcxproj]
~~~

traversaro, 2020-03-16:
The error may seem enigmatic, but it is not with a bit of info on how the linking process works on Windows/MSVC. The `.lib` file is just a stub that contains information of the exported symbols of a given shared (i.e. `.dll`) library. If a library does not export any symbol, the `.lib` is not created at all, resulting in this error. See https://github.com/robotology/unicycle-footstep-planner/issues/12 for a related issue.

traversaro, 2020-03-16:
The version that does not compile on Windows has been released as 0.3.0, see https://github.com/robotology/walking-controllers/commit/57f9d6b5e3c2ce4b020c4bf65f7768182d85d775 .

traversaro, 2020-03-17:
@GiulioRomualdi Do you want me to handle this. Add the definition of  `CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS` per https://github.com/robotology/how-to-export-cpp-library/blob/bc184877b002737d457f773fbbcac3bbef4208d3/CMakeLists.txt#L44 should be sufficient.

GiulioRomualdi, 2020-03-17:
I can handle this

traversaro, 2020-03-17:
> 
> 
> I can handle this

Thanks!

closed 2020-03-17 by GiulioRomualdi

traversaro, 2020-03-17:
Unfortunately the compilation is still failing, see https://github.com/robotology/robotology-superbuild/runs/513826894?check_suite_focus=true . 
The error in particular is: 
~~~
2020-03-17T14:16:10.7423498Z D:\a\robotology-superbuild\robotology-superbuild\robotology\walking-controllers\src\TimeProfiler\src\TimeProfiler.cpp(96,24): error C2039: 'to_string': is not a member of 'std' [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\walking-controllers\src\TimeProfiler\TimeProfiler.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\walking-controllers.vcxproj]
2020-03-17T14:16:10.7424290Z   C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.24.28314\include\map(25): message : see declaration of 'std' [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\walking-controllers\src\TimeProfiler\TimeProfiler.vcxproj]
2020-03-17T14:16:10.7433262Z D:\a\robotology-superbuild\robotology-superbuild\robotology\walking-controllers\src\TimeProfiler\src\TimeProfiler.cpp(96,33): error C3861: 'to_string': identifier not found [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\walking-controllers\src\TimeProfiler\TimeProfiler.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\walking-controllers.vcxproj]
~~~ 

At a first glance, the problem seems in the `TimeProfiler.cpp` uses `std::to_string` without including the correct header (see https://github.com/robotology/walking-controllers/blob/eaac283a9b1e7b3275f853b9d9bef67b73136b80/src/TimeProfiler/src/TimeProfiler.cpp#L96), violating IWYU. Compilation works fine on *nix probably because the string header is transitively included by some other header.

GiulioRomualdi, 2020-03-20:
Probably I should use [`IWYU`](https://github.com/include-what-you-use/include-what-you-use/blob/master/README.md) in the future

PR #59 (2020-03-17): Update the main CMakeLists.txt file to be compatible with windows
Fix #58 

After this PR a new version of the code will be released