gbionics/walking-controllers #83: Closing the Walking Module may crash the yarprbotinterface

S-Dafarra, 2021-03-05:
This happens quite seldom, but it happens.

I wonder what we can do at the application level that can ultimately lead the ``yarprobotinterface`` to crash.

cc @traversaro @GiulioRomualdi

S-Dafarra, 2021-03-26:
It just happened once in simulation that WBD crashed after closing the ``WalkingModule``. I guess this can cause in turn ``yarprobotinterface`` to crash. Here the output of the terminal:
```
[INFO] run phase starting...
[DEBUG] yarprobotinterface running happily
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_front/cartesianEndEffectorWrench:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_rear/cartesianEndEffectorWrench:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_front/cartesianEndEffectorWrench:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_rear/cartesianEndEffectorWrench:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_front/cartesianEndEffectorWrench:i
[INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /wholeBodyDynamics/left_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_rear/cartesianEndEffectorWrench:i
[INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_front/cartesianEndEffectorWrench:i
free(): double free detected in tcache 2
Aborted (core dumped)
```

S-Dafarra, 2021-03-26:
I have launched WBD with ``valgrind`` and there seems to be a lot memory leaks:
```
sdafarra@iiticublap104:~$ YARP_CLOCK=/clock valgrind yarprobotinterface --config launch-wholebodydynamics.xml
==220691== Memcheck, a memory error detector
==220691== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==220691== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==220691== Command: yarprobotinterface --config launch-wholebodydynamics.xml
==220691== 
[INFO] |yarp.os.Port| Port /iiticublap104/yarprobotinterface/220691/clock:i active at tcp://10.240.9.61:10274/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /clock to /iiticublap104/yarprobotinterface/220691/clock:i using tcp
[ERROR] |yarp.os.Property| cannot read from yarprobotinterface.ini
[DEBUG] Reading file /home/sdafarra/Software/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml
[WARNING] Invalid syntax while loading /home/sdafarra/Software/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml at line 1 . Expected document of type robot . Found devices
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /home/sdafarra/Software/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml
[DEBUG] Preprocessor complete in:  0.077 s
[WARNING] Invalid syntax while loading  at line 4 . "robot" element should contain the "portprefix" attribute. Using "name" attribute
[INFO] |yarp.os.Port| Port /iCubGazeboV3/yarprobotinterface active at tcp://10.240.9.61:10053/
[INFO] startup phase starting...
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/torso/rpc:o active at tcp://10.240.9.61:10054/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/torso/command:o active at tcp://10.240.9.61:10055/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/torso/stateExt:i active at tcp://10.240.9.61:10056/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/torso/rpc:o to /icubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/torso/command:o to /icubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_arm/rpc:o active at tcp://10.240.9.61:10057/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_arm/command:o active at tcp://10.240.9.61:10058/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_arm/stateExt:i active at tcp://10.240.9.61:10059/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_arm/rpc:o to /icubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_arm/command:o to /icubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/left_arm/stateExt:o to /wholeBodyDynamics/left_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_arm/rpc:o active at tcp://10.240.9.61:10060/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_arm/command:o active at tcp://10.240.9.61:10061/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_arm/stateExt:i active at tcp://10.240.9.61:10062/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_arm/rpc:o to /icubSim/right_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_arm/command:o to /icubSim/right_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/right_arm/stateExt:o to /wholeBodyDynamics/right_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_leg/rpc:o active at tcp://10.240.9.61:10063/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_leg/command:o active at tcp://10.240.9.61:10064/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_leg/stateExt:i active at tcp://10.240.9.61:10065/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_leg/rpc:o to /icubSim/left_leg/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_leg/command:o to /icubSim/left_leg/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/left_leg/stateExt:o to /wholeBodyDynamics/left_leg/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_leg/rpc:o active at tcp://10.240.9.61:10066/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_leg/command:o active at tcp://10.240.9.61:10067/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_leg/stateExt:i active at tcp://10.240.9.61:10068/
[... cut ...]
==220691==                       of which reachable via heuristic:
==220691==                         stdstring          : 40 bytes in 1 blocks
==220691==                         multipleinheritance: 576 bytes in 4 blocks
==220691==         suppressed: 0 bytes in 0 blocks
==220691== Rerun with --leak-check=full to see details of leaked memory
==220691== 
==220691== For lists of detected and suppressed errors, rerun with: -s
==220691== ERROR SUMMARY: 37 errors from 31 contexts (suppressed: 6 from 2)
Aborted (core dumped)
```

S-Dafarra, 2021-03-26:
I think that ``getOutputCount``, used in https://github.com/robotology/whole-body-estimators/blob/2ffe2d0c65a70b5435a4aa97882ab1857cb091a0/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L2497, is indeed the cause. I suspect that this is due to the ``cleanUnits`` method, used in https://github.com/robotology/yarp/blob/93898dfbcbed448b3b0a01b78ae229f7b5731199/src/libYARP_os/src/yarp/os/impl/PortCore.cpp#L1428 that may cause a double segfault when the port is actually closing.

S-Dafarra, 2021-03-26:
I am opening an issue on ``yarp``.

S-Dafarra, 2021-03-26:
Here https://github.com/robotology/yarp/issues/2531

traversaro, 2021-03-26:
Thanks for opening the issue on YARP, I wonder if this could be related to some of the old recurring crashing test (see https://github.com/robotology/yarp/issues/880). However, at the WBD level, I wonder what is the point of having `_port.getOutputCount()` in https://github.com/robotology/whole-body-estimators/blob/2ffe2d0c65a70b5435a4aa97882ab1857cb091a0/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L2497 at all. 

This is a relic from the old wbd (see https://github.com/robotology/icub-main/blob/65366322b10adeaa52f94acc44411d35bb3b6988/src/modules/wholeBodyDynamics/observerThread.cpp#L1161) that I always copied in all the evolutions of WBD, in a rather [cargo-cultish](https://en.wikipedia.org/wiki/Cargo_cult_programming) way. If `getOutputCount` is (as it appears from https://github.com/robotology/yarp/blob/9b52a71e176e23ebe42956cf5b23d8ae40b255f0/src/libYARP_os/src/yarp/os/impl/PortCore.cpp#L615) contacting all the connected ports to check if they are still alive, then it probably slows down the execution much more than any benefit that you could get by skipping the `_port.prepare()/_port.write()` calls. 

Actually it seems that this pattern is quite widespared across icub-main, I will open a discussion  in Community to discuss that.

S-Dafarra, 2021-03-26:
I will open a PR to remove it.

traversaro, 2021-03-26:
> Actually it seems that this pattern is quite widespared across icub-main, I will open a discussion in Community to discuss that.

See https://github.com/robotology/yarp/discussions/2533 .

> I will open a PR to remove it.

Thanks, even if from the discussion in https://github.com/robotology/yarp/discussions/2533 it turns out that `_port.getOutputCount()` is inexpensive, I think that in such realtime-oriented code we should prefer determinism over saving computational time based on external conditions.

prashanthr05, 2021-03-26:
> It just happened once in simulation that WBD crashed after closing the `WalkingModule`. I guess this can cause in turn `yarprobotinterface` to crash. Here the output of the terminal:
> 
> ```
> [INFO] run phase starting...
> [DEBUG] yarprobotinterface running happily
> [INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_front/cartesianEndEffectorWrench:i using tcp
> [INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_rear/cartesianEndEffectorWrench:i using tcp
> [INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_front/cartesianEndEffectorWrench:i using tcp
> [INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_rear/cartesianEndEffectorWrench:i using tcp
> [INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_front/cartesianEndEffectorWrench:i
> [INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /wholeBodyDynamics/left_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_rear/cartesianEndEffectorWrench:i
> [INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_front/cartesianEndEffectorWrench:i
> free(): double free detected in tcache 2
> Aborted (core dumped)
> ```
> 
> cc @prashanthr05

I am trying to remember if I have ever faced such an issue before, but in vain. I have only faced with issues reading the sensors, but with the network load of another device (base estimation) trying to read parallely the sensor buffers. Sometimes, it happens so that one of the two devices can't read from the buffers at all anymore.

But a crash seems new to me.

S-Dafarra, 2021-03-26:
Removing ``getOutputCount`` seems to remove all the memory leaks.

S-Dafarra, 2021-03-26:
I was too fast, there are still some, but it is an invalid read/write
```
[INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /wholeBodyDynamics/right_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_rear/cartesianEndEffectorWrench:i
==389301== Thread 69:
==389301== Invalid read of size 4
==389301==    at 0x5054EB8: ACE_SOCK::close() (in /usr/lib/libACE-6.4.5.so)
==389301==    by 0x49DBC66: non-virtual thunk to yarp::os::impl::SocketTwoWayStream::interrupt() (SocketTwoWayStream.h:93)
==389301==    by 0x49D8955: interrupt (Protocol.cpp:261)
==389301==    by 0x49D8955: yarp::os::impl::Protocol::interrupt() (Protocol.cpp:251)
==389301==    by 0x49D5C57: yarp::os::impl::PortCoreOutputUnit::closeMain() (PortCoreOutputUnit.cpp:225)
==389301==    by 0x49AD52A: yarp::os::impl::PortCore::reapUnits() (PortCore.cpp:582)
==389301==    by 0x49B52D1: yarp::os::impl::PortCore::run() (PortCore.cpp:243)
==389301==    by 0x49E546C: theExecutiveBranch(void*) (ThreadImpl.cpp:84)
==389301==    by 0x4B4AD83: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==389301==    by 0x50EA608: start_thread (pthread_create.c:477)
==389301==    by 0x4D92292: clone (clone.S:95)
==389301==  Address 0x5f2dc18 is 24 bytes inside a block of size 120 free'd
==389301==    at 0x483D1CF: operator delete(void*, unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==389301==    by 0x4966B13: yarp::os::ShiftStream::Private::close() (ShiftStream.cpp:40)
==389301==    by 0x49D905E: yarp::os::impl::Protocol::closeHelper() (Protocol.cpp:538)
==389301==    by 0x49D59FD: yarp::os::impl::PortCoreOutputUnit::closeBasic() (PortCoreOutputUnit.cpp:207)
==389301==    by 0x49D6046: yarp::os::impl::PortCoreOutputUnit::sendHelper() (PortCoreOutputUnit.cpp:335)
==389301==    by 0x49D6795: yarp::os::impl::PortCoreOutputUnit::run() (PortCoreOutputUnit.cpp:97)
==389301==    by 0x49E546C: theExecutiveBranch(void*) (ThreadImpl.cpp:84)
==389301==    by 0x4B4AD83: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==389301==    by 0x50EA608: start_thread (pthread_create.c:477)
==389301==    by 0x4D92292: clone (clone.S:95)
==389301==  Block was alloc'd at
==389301==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==389301==    by 0x49E3A7B: yarp::os::impl::TcpFace::write(yarp::os::Contact const&) (TcpFace.cpp:125)
==389301==    by 0x4917721: yarp::os::Carriers::connect(yarp::os::Contact const&) (Carriers.cpp:300)
==389301==    by 0x49B5BA4: yarp::os::impl::PortCore::addOutput(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void*, yarp::os::OutputStream*, bool) (PortCore.cpp:967)
==389301==    by 0x49B744A: operator() (PortCore.cpp:1901)
==389301==    by 0x49B744A: yarp::os::impl::PortCore::adminBlock(yarp::os::ConnectionReader&, void*) (PortCore.cpp:2609)
==389301==    by 0x49D328C: yarp::os::impl::PortCoreInputUnit::run() (PortCoreInputUnit.cpp:313)
==389301==    by 0x49E546C: theExecutiveBranch(void*) (ThreadImpl.cpp:84)
==389301==    by 0x4B4AD83: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==389301==    by 0x50EA608: start_thread (pthread_create.c:477)
==389301==    by 0x4D92292: clone (clone.S:95)
==389301== 
==389301== Invalid write of size 1
==389301==    at 0x49D8956: interrupt (Protocol.cpp:262)
==389301==    by 0x49D8956: yarp::os::impl::Protocol::interrupt() (Protocol.cpp:251)
==389301==    by 0x49D5C57: yarp::os::impl::PortCoreOutputUnit::closeMain() (PortCoreOutputUnit.cpp:225)
==389301==    by 0x49AD52A: yarp::os::impl::PortCore::reapUnits() (PortCore.cpp:582)
==389301==    by 0x49B52D1: yarp::os::impl::PortCore::run() (PortCore.cpp:243)
==389301==    by 0x49E546C: theExecutiveBranch(void*) (ThreadImpl.cpp:84)
==389301==    by 0x4B4AD83: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==389301==    by 0x50EA608: start_thread (pthread_create.c:477)
==389301==    by 0x4D92292: clone (clone.S:95)
==389301==  Address 0x76f7d30 is 48 bytes inside a block of size 392 free'd
==389301==    at 0x483D1CF: operator delete(void*, unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==389301==    by 0x49D5A0F: yarp::os::impl::PortCoreOutputUnit::closeBasic() (PortCoreOutputUnit.cpp:208)
==389301==    by 0x49D6046: yarp::os::impl::PortCoreOutputUnit::sendHelper() (PortCoreOutputUnit.cpp:335)
==389301==    by 0x49D6795: yarp::os::impl::PortCoreOutputUnit::run() (PortCoreOutputUnit.cpp:97)
==389301==    by 0x49E546C: theExecutiveBranch(void*) (ThreadImpl.cpp:84)
==389301==    by 0x4B4AD83: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==389301==    by 0x50EA608: start_thread (pthread_create.c:477)
==389301==    by 0x4D92292: clone (clone.S:95)
==389301==  Block was alloc'd at
==389301==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==389301==    by 0x49E3D23: yarp::os::impl::TcpFace::write(yarp::os::Contact const&) (TcpFace.cpp:145)
==389301==    by 0x4917721: yarp::os::Carriers::connect(yarp::os::Contact const&) (Carriers.cpp:300)
==389301==    by 0x49B5BA4: yarp::os::impl::PortCore::addOutput(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void*, yarp::os::OutputStream*, bool) (PortCore.cpp:967)
==389301==    by 0x49B744A: operator() (PortCore.cpp:1901)
==389301==    by 0x49B744A: yarp::os::impl::PortCore::adminBlock(yarp::os::ConnectionReader&, void*) (PortCore.cpp:2609)
==389301==    by 0x49D328C: yarp::os::impl::PortCoreInputUnit::run() (PortCoreInputUnit.cpp:313)
==389301==    by 0x49E546C: theExecutiveBranch(void*) (ThreadImpl.cpp:84)
==389301==    by 0x4B4AD83: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==389301==    by 0x50EA608: start_thread (pthread_create.c:477)
==389301==    by 0x4D92292: clone (clone.S:95)
==389301== 
```

traversaro, 2021-03-26:
> 
> 
> Removing `getOutputCount` seems to remove all the memory leaks.

Nitpick: invalid access memory (i.e. reading memory not initialized or not allocated) technically speaking are not memory leaks.

S-Dafarra, 2021-03-26:
> Nitpick: invalid access memory (i.e. reading memory not initialized or not allocated) technically speaking are not memory leaks.

What about double free? :grin:

traversaro, 2021-03-26:
Not even those. Memory leak is when you call a new/malloc, and you never call delete/free. Double free is the exact opposite! :D

S-Dafarra, 2021-04-01:
I tested the new version today on the robot and seems to be working fine. Closing this. If it happens again we will keep investigating.

closed 2021-04-01 by S-Dafarra