gbionics/walking-controllers #84: The robot may start oscillating visibly right after starting the walking

S-Dafarra, 2021-03-05:
We experienced this during a demo with the robot that started to oscillate visibly from left to right. It would nice to investigate possible causes and solutions.

cc @isorrentino

isorrentino, 2021-03-22:
The last week I launched the walking several time and I acquired data for each one. I used https://github.com/robotology/walking-controllers#how-to-dump-data to log data. I cannot upload the datasets as the file is too big.
Here the datasets:
[walkingStartupData.zip](https://github.com/robotology/walking-controllers/files/6180542/walkingStartupData.zip)

I still need to analyze them, but in general, I can say that the robot did not repeat the behaviour we had in Milan. Just once it started oscillating but much much less and actually I'm not sure I can see something from the data.

isorrentino, 2021-04-01:
From these datasets, I do not see anything strange. All the measured quantities are ok. I'll try to do other experiments by pushing the robot to see if it replicates the behavior.

isorrentino, 2021-07-19:
Whether the behaviour will newly appear, I will open a new issue for further investigation. Closing.

closed 2021-07-19 by isorrentino