gbionics/walking-controllers #95: `WalkingLoggerModule` is not running properly

mebbaid, 2021-11-19:
When trying to use the `WalkingLoggerModule` I get the following error
```

[INFO] [RPC Server] The following data will be stored:  time dcm_x dcm_y dcm_des_x dcm_des_y dcm_des_dx dcm_des_dy zmp_x zmp_y zmp_des_x zmp_des_y com_x com_y com_z com_des_x com_des_y com_des_z com_des_dx com_des_dy com_des_dz lf_x lf_y lf_z lf_roll lf_pitch lf_yaw rf_x rf_y rf_z rf_roll rf_pitch rf_yaw lf_des_x lf_des_y lf_des_z lf_des_roll lf_des_pitch lf_des_yaw rf_des_x rf_des_y rf_des_z rf_des_roll rf_des_pitch rf_des_yaw torso_pitch torso_roll torso_yaw l_shoulder_pitch l_shoulder_roll l_shoulder_yaw l_elbow l_wrist_prosup l_wrist_pitch l_wrist_yaw r_shoulder_pitch r_shoulder_roll r_shoulder_yaw r_elbow r_wrist_prosup r_wrist_pitch r_wrist_yaw l_hip_pitch l_hip_roll l_hip_yaw l_knee l_ankle_pitch l_ankle_roll r_hip_pitch r_hip_roll r_hip_yaw r_knee r_ankle_pitch r_ankle_roll torso_pitch_des torso_roll_des torso_yaw_des l_shoulder_pitch_des l_shoulder_roll_des l_shoulder_yaw_des l_elbow_des l_wrist_prosup_des l_wrist_pitch_des l_wrist_yaw_des r_shoulder_pitch_des r_shoulder_roll_des r_shoulder_yaw_des r_elbow_des r_wrist_prosup_des r_wrist_pitch_des r_wrist_yaw_des l_hip_pitch_des l_hip_roll_des l_hip_yaw_des l_knee_des l_ankle_pitch_des l_ankle_roll_des r_hip_pitch_des r_hip_roll_des r_hip_yaw_des r_knee_des r_ankle_pitch_des r_ankle_roll_des torso_pitch_ret torso_roll_ret torso_yaw_ret l_shoulder_pitch_ret l_shoulder_roll_ret l_shoulder_yaw_ret l_elbow_ret l_wrist_prosup_ret l_wrist_pitch_ret l_wrist_yaw_ret r_shoulder_pitch_ret r_shoulder_roll_ret r_shoulder_yaw_ret r_elbow_ret r_wrist_prosup_ret r_wrist_pitch_ret r_wrist_yaw_ret l_hip_pitch_ret l_hip_roll_ret l_hip_yaw_ret l_knee_ret l_ankle_pitch_ret l_ankle_roll_ret r_hip_pitch_ret r_hip_roll_ret r_hip_yaw_ret r_knee_ret r_ankle_pitch_ret r_ankle_roll_ret 
[ERROR] [updateModule] The size of the vector is not the one expected. Expected:  130  received:  112
[INFO] |yarp.os.RFModule| RFModule closing.

```
This error seems to be because of 85cdd95ad79c079e4443b714d417b28f62702dc3.

The temporary workaround, as suggested by @S-Dafarra  is to `checkout` back to before [this change](https://github.com/robotology/walking-controllers/commit/85cdd95ad79c079e4443b714d417b28f62702dc3) to be able to log and plot data properly. However this issue is to document the issue.

S-Dafarra, 2021-11-19:
cc @GiulioRomualdi

GiulioRomualdi, 2021-11-22:
Hi @mebbaid, this is actually an intrinsic problem of the implementation of the WalkingLogger. Now I try to explain to you why this is happening. The explanation is based on the latest devel commit: https://github.com/robotology/walking-controllers/commit/9d8dde45159e43187d672ad9bf4991734714d7b7.

The WalkingLogger is an application that you run and the WalkingController established the connection if the `dump_data` parameter is set to `1`.

If you check the WalkingLogger code you will see that the logger is actually composed of a `WalkingLoggerServer` (this is the application that you run) and the `WalkingLoggerClient` (this is the library that you load in the `WalkingModule`)

To start logging the parameter you should call `startRecord` method. The first parameter of the logger is `record` and then the list of each element of the vector that will be saved. This is required to define the  first row of the csv table

https://github.com/robotology/walking-controllers/blob/9d8dde45159e43187d672ad9bf4991734714d7b7/src/WalkingModule/src/Module.cpp#L1354-L1384

In this case, the table is composed of the following columns

| `dcm_x` |  `dcm_y` | `dcm_z` | ..... | `r_ankle_pitch_ret` | `r_ankle_roll_ret`|
| --------|--------|--------|--------|--------|--------| 

As you can see the code has been committed in status such that we log 29 robot joints and 29 retargeting joints (this is the status in which we log the data if the retargeting is enabled)

To log the data you send the data to logger you should call `sendData`

https://github.com/robotology/walking-controllers/blob/9d8dde45159e43187d672ad9bf4991734714d7b7/src/WalkingModule/src/Module.cpp#L887-L897

as you can see here the robot joint state the desired joint state and the retargeting joint states are sent using the following vectors
```cpp
m_robotControlHelper->getJointPosition(),
m_qDesired,                                     m_retargetingClient->jointValues()
```

If you run the walking with the joypad the dimension of the three vectors is equal to `23` each.  

`sendData` is a template variadic function that is in charge to attach all the vectors together in a long `yarp::os::Vector` and send it to the logger.
As a consequence, if the number of the elements is defined [here](https://github.com/robotology/walking-controllers/blob/9d8dde45159e43187d672ad9bf4991734714d7b7/src/WalkingModule/src/Module.cpp#L1354-L1384) is different from the sum of the size of the all vectors used [here](https://github.com/robotology/walking-controllers/blob/9d8dde45159e43187d672ad9bf4991734714d7b7/src/WalkingModule/src/Module.cpp#L887-L897) you will have an error. 

I hope that the problem is now clear. Now let's move to the possible solutions:
1. Commit a new version of the code in which we remove some joints [here](https://github.com/robotology/walking-controllers/blob/9d8dde45159e43187d672ad9bf4991734714d7b7/src/WalkingModule/src/Module.cpp#L1354-L1384) to make everything work: you can do this in your local branch but it does not make sense to push it, indeed you will be able to log the data if you run the walking with the joypad but not with the xsens/oculus.
2. Restructure the WalkingLogger such that the size is automatically handled by configuration files: this could be a solution but since we have to spend time to implement it I would try to move the effort to something better than a logger that I wrote during the master thesis 😃 
3. Try to adapt the [`YarpRobotLoggerDevice`](https://github.com/ami-iit/bipedal-locomotion-framework/tree/master/devices/YarpRobotLoggerDevice) in order to accept also values data from the walking: I like this solution. It is more complex but more modular. What we can do in this case is to define yarp devices that can be used to stream matrices/vectors using thrift messages. You can check [here](https://github.com/GiulioRomualdi/YarpMatLogger/blob/master/src/LibCustomMsg/thrifts/MatrixCollection/MatrixCollection.thrift) for an example. This device can be used to collect data in the `YarpRobotLoggerDevice` (or also in the [`YarpTelemetryDevice`](https://github.com/robotology/yarp-telemetry)) or also to visualize them on what [@nunoguedelha is implementing](https://github.com/ami-iit/yarp-openmct)

If you want to try to solve this problem it would be wonderful, you can schedule a meeting and we can discuss it together

cc @isorrentino

GiulioRomualdi, 2021-11-22:
cc @traversaro

mebbaid, 2021-11-22:
@GiulioRomualdi thanks for the clear explanation of the problem and the suggested solutions. Indeed for the time being I am working on a different local branch to be able to plot.

Concerning the suggested solutions, I would agree with your advice on proceeding with option (3) not least because you know better the "optimal" solution to the problem, and also for the selfish reason of wanting to get involved more with implementation aspects. 

I can organize a meeting for this Friday together with you and the team members tagged if that is OK.

isorrentino, 2021-11-23:
Great explanation @GiulioRomualdi. I would proceed with the third option :)

traversaro, 2021-11-26:
> 3\. (or also in the [`YarpTelemetryDevice`](https://github.com/robotology/yarp-telemetry))

I guess you refer to the [`telemetryDeviceDumper`](https://github.com/robotology/yarp-telemetry#telemetrydevicedumper) here, right? xref https://github.com/ami-iit/bipedal-locomotion-framework/issues/189

lrapetti, 2022-02-11:
If I am not wrong solution 3 from https://github.com/robotology/walking-controllers/issues/95#issuecomment-975332410 was implemented right? We probably can close this issue.

mebbaid, 2022-02-11:
yes, indeed this issue was solved with this [YarpLoggerDevice](https://github.com/ami-iit/bipedal-locomotion-framework/pull/487). Closing

closed 2022-02-11 by mebbaid