gbionics/walking-controllers #102: Missing dependency on icub-main in the documentation

lrapetti, 2022-02-04:
In the repo documentation (https://github.com/robotology/walking-controllers#page_facing_up-dependencies), [`icub-main`](https://github.com/robotology/icub-main) is not mentioned, but it seems to be a required dependency.

cc @SimoneMic feel free to add here any detail

SimoneMic, 2022-02-07:
Yeah everything fine, did all the installation in https://icub-tech-iit.github.io/documentation/sw_installation/linux_from_sources_manual/

lrapetti, 2022-02-10:
The dependency on [`whole-body-estimators`](https://github.com/robotology/whole-body-estimators) is missing as well in case you want to run the simulation example (https://github.com/robotology/walking-controllers#computer-how-to-run-the-simulation), we could specify it in the documentation (also [`icub-models`](https://github.com/robotology/icub-models) would be needed for running the example with iCub).