gbionics/walking-controllers #107: iCubGazebo_V3 Falls Frequently in Simulation on Gazebo

SimoneMic, 2022-02-14:
Hello everybody!

I am using in simulation (Gazebo) the iCubGazebo_V3 (no hands) model and it keeps falling after a little while. The best I could do is 4 meters in a straight line. I have followed step-by-step the instructions in https://github.com/robotology/walking-controllers#computer-how-to-run-the-simulation and I'm controlling it via `yarp rpc /walking-coordinator/rpc` using the command `setGoal x y`.

I have also tried to wait the robot to reach the setGoal location and to relaunch the command, like a stop-and-go, but it never manages to go after the third command that it falls. So it would seem that in continuous walking it behaves better.

I have also used the pid parameters suggested by @mebbaid in the following link: https://github.com/robotology/icub-models/pull/126/files but to no avail.

I am using the latest build of walking-controllers (master).

Here I attach the terminal output of the walking module. 

Maybe I am missing something that I can't see.

Thank you

```
simone@IITICUBLAP249:~$ YARP_CLOCK=/clock WalkingModule
[INFO] |yarp.os.Port|/IITICUBLAP249/WalkingModule/18816/clock:i| Port /IITICUBLAP249/WalkingModule/18816/clock:i active at tcp://127.0.0.1:10094/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP249/WalkingModule/18816/clock:i| Receiving input from /clock to /IITICUBLAP249/WalkingModule/18816/clock:i using tcp
[DEBUG] |yarp.dev.PolyDriver|remotecontrolboardremapper| Parameters are (REMOTE_CONTROLBOARD_OPTIONS (writeStrict on)) (axesNames (torso_pitch torso_roll torso_yaw l_shoulder_pitch l_shoulder_roll l_shoulder_yaw l_elbow r_shoulder_pitch r_shoulder_roll r_shoulder_yaw r_elbow l_hip_pitch l_hip_roll l_hip_yaw l_knee l_ankle_pitch l_ankle_roll r_hip_pitch r_hip_roll r_hip_yaw r_knee r_ankle_pitch r_ankle_roll)) (device remotecontrolboardremapper) (localPortPrefix "/walking-coordinator/remoteControlBoard") (remoteControlBoards ("/icubSim/torso" "/icubSim/left_arm" "/icubSim/right_arm" "/icubSim/left_leg" "/icubSim/right_leg"))
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/icubSim/torso") (remote "/icubSim/torso") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o| Port /walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o active at tcp://127.0.0.1:10095/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/torso/command:o| Port /walking-coordinator/remoteControlBoard/icubSim/torso/command:o active at tcp://127.0.0.1:10096/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/torso/stateExt:i| Port /walking-coordinator/remoteControlBoard/icubSim/torso/stateExt:i active at tcp://127.0.0.1:10097/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o to /icubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/torso/command:o| Sending output from /walking-coordinator/remoteControlBoard/icubSim/torso/command:o to /icubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/icubSim/torso/stateExt:i| Receiving input from /icubSim/torso/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/icubSim/left_arm") (remote "/icubSim/left_arm") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/left_arm/rpc:o| Port /walking-coordinator/remoteControlBoard/icubSim/left_arm/rpc:o active at tcp://127.0.0.1:10098/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o| Port /walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o active at tcp://127.0.0.1:10099/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/left_arm/stateExt:i| Port /walking-coordinator/remoteControlBoard/icubSim/left_arm/stateExt:i active at tcp://127.0.0.1:10100/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_arm/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/icubSim/left_arm/rpc:o to /icubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o| Sending output from /walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o to /icubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_arm/stateExt:i| Receiving input from /icubSim/left_arm/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/left_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/icubSim/right_arm") (remote "/icubSim/right_arm") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/right_arm/rpc:o| Port /walking-coordinator/remoteControlBoard/icubSim/right_arm/rpc:o active at tcp://127.0.0.1:10101/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o| Port /walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o active at tcp://127.0.0.1:10102/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/right_arm/stateExt:i| Port /walking-coordinator/remoteControlBoard/icubSim/right_arm/stateExt:i active at tcp://127.0.0.1:10103/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_arm/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/icubSim/right_arm/rpc:o to /icubSim/right_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o| Sending output from /walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o to /icubSim/right_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_arm/stateExt:i| Receiving input from /icubSim/right_arm/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/right_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (device remote_controlboard) (local "/walking-coordinator/remoteControlBoard/icubSim/left_leg") (remote "/icubSim/left_leg") (writeStrict on)
[INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/left_leg/rpc:o| Port /walking-coordinator/remoteControlBoard/icubSim/left_leg/rpc:o active at tcp://127.0.0.1:10104/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/left_leg/command:o| Port /walking-coordinator/remoteControlBoard/icubSim/left_leg/command:o active at tcp://127.0.0.1:10105/
[INFO] |yarp.os.Port|/walking-coordinator/remoteControlBoard/icubSim/left_leg/stateExt:i| Port /walking-coordinator/remoteControlBoard/icubSim/left_leg/stateExt:i active at tcp://127.0.0.1:10106/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_leg/rpc:o| Sending output from /walking-coordinator/remoteControlBoard/icubSim/left_leg/rpc:o to /icubSim/left_leg/rpc:i using tcp
[... cut ...]
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_leg/stateExt:i| Removing input from /icubSim/right_leg/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/right_leg/stateExt:i
[INFO] |yarp.os.RFModule| RFModule finished.
[WARNING] |yarp.os.NetworkClock| Destroying network clock
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP249/WalkingModule/18816/clock:i| Removing input from /clock to /IITICUBLAP249/WalkingModule/18816/clock:i
```
[image]

cc @lrapetti @randaz81

mebbaid, 2022-02-14:
ciao @SimoneMic , are you available tomorrow in IIT (preferably after 14:00) ? so that we can compare our set-ups to see if there is any difference and try to debug/iterate.

SimoneMic, 2022-02-14:
> ciao @SimoneMic , are you available tomorrow in IIT for a f2f (preferably after 14:00) ? so that we can compare our set-ups to see if there is any difference and try to debug/iterate.

Unfortunately this week I'm not in Genova, but the next one I'll be there.

mebbaid, 2022-02-14:
ok, we can arrange a ms-teams meeting tomorrow, because (at least to me, others can be more helpful here) it is difficult to debug without going through the set-up thoroughly.

SimoneMic, 2022-02-14:
Ok, thanks, fine by me.

mebbaid, 2022-02-15:
> Other information: I have the latest versions of walking-controllers, unicycle footstep planner. Also I have gazebo 11 and the latest version of YARP (3.6).

yeah, I should start by saying that, apart from using the `super-build` , the individual relevant repositories were aligned in both our machines. Alas, I was not able to reproduce his errors and almost constant falls.

lrapetti, 2022-02-16:
@SimoneMic probably a video of the failure might also be of help for debugging your issue

SimoneMic, 2022-02-16:
Here's a video 
[iCubGazebo_Fall.zip](https://github.com/robotology/walking-controllers/files/8078398/iCubGazebo_Fall.zip)

Let me know if you need something else.

SimoneMic, 2022-02-16:
I've also tried to run the configuration presented in https://github.com/robotology/walking-controllers/blob/fc74f101e1e933ac6b89e28ca52559ef18d4cd9a/src/WalkingModule/app/robots/iCubGenova09/dcm_walking/joypad_control/robotControl.ini#L21-L25
running the ankle joints in compliant mode (as suggested by @lrapetti ). But the robot fell forward right after the `startWalking` command.
Here's the log:
```
q desired IK   18.5369  0.236021  -0.05871  -1.91926   16.7715  -12.6683   39.7101  -1.95081    16.789  -12.6609   39.7137   2.59106  0.623582 -0.289821  -58.3052  -33.6794 -0.731743   2.62289  0.494828 -0.223192  -58.3616  -33.7034 -0.498707
[INFO] [WalkingModule::updateModule] The robot is prepared.
IK: 1.475400 ms Total: 2.669100 ms 
IK: 0.965000 ms Total: 2.098400 ms 
IK: 1.008600 ms Total: 2.024900 ms 
IK: 1.129500 ms Total: 2.342200 ms 
IK: 0.954300 ms Total: 1.771800 ms 
IK: 0.814500 ms Total: 1.600900 ms 
IK: 1.162400 ms Total: 2.243900 ms 
IK: 0.945400 ms Total: 1.495000 ms 
IK: 1.126000 ms Total: 2.373200 ms 
IK: 0.799600 ms Total: 1.718600 ms 
IK: 0.978500 ms Total: 1.764800 ms 
IK: 1.428200 ms Total: 2.525300 ms 
[ERROR] [RobotInterface::setDirectPositionReferences] The worst error between the current and the  desired position of the  l_shoulder_roll  joint is greater than 0.5 rad.
[ERROR] [WalkingModule::updateModule] Error while setting the reference position to iCub.
[INFO] |yarp.os.RFModule| RFModule closing.
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/left_foot_front/cartesianEndEffectorWrench:i| Removing input from /wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_front/cartesianEndEffectorWrench:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/left_foot_rear/cartesianEndEffectorWrench:i| Removing input from /wholeBodyDynamics/left_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/left_foot_rear/cartesianEndEffectorWrench:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/right_foot_front/cartesianEndEffectorWrench:i| Removing input from /wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_front/cartesianEndEffectorWrench:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/right_foot_rear/cartesianEndEffectorWrench:i| Removing input from /wholeBodyDynamics/right_foot_rear/cartesianEndEffectorWrench:o to /walking-coordinator/right_foot_rear/cartesianEndEffectorWrench:i
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o| Removing output from /walking-coordinator/remoteControlBoard/icubSim/torso/rpc:o to /icubSim/torso/rpc:i
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/torso/command:o| output for route /walking-coordinator/remoteControlBoard/icubSim/torso/command:o->udp->/icubSim/torso/command:i asking other side to close by out-of-band means
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/torso/command:o| Removing output from /walking-coordinator/remoteControlBoard/icubSim/torso/command:o to /icubSim/torso/command:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/icubSim/torso/stateExt:i| Removing input from /icubSim/torso/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/torso/stateExt:i
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_arm/rpc:o| Removing output from /walking-coordinator/remoteControlBoard/icubSim/left_arm/rpc:o to /icubSim/left_arm/rpc:i
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o| output for route /walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o->udp->/icubSim/left_arm/command:i asking other side to close by out-of-band means
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o| Removing output from /walking-coordinator/remoteControlBoard/icubSim/left_arm/command:o to /icubSim/left_arm/command:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_arm/stateExt:i| Removing input from /icubSim/left_arm/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/left_arm/stateExt:i
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_arm/rpc:o| Removing output from /walking-coordinator/remoteControlBoard/icubSim/right_arm/rpc:o to /icubSim/right_arm/rpc:i
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o| output for route /walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o->udp->/icubSim/right_arm/command:i asking other side to close by out-of-band means
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o| Removing output from /walking-coordinator/remoteControlBoard/icubSim/right_arm/command:o to /icubSim/right_arm/command:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_arm/stateExt:i| Removing input from /icubSim/right_arm/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/right_arm/stateExt:i
[INFO] |yarp.os.impl.PortCoreOutputUnit|/walking-coordinator/remoteControlBoard/icubSim/left_leg/rpc:o| Removing output from /walking-coordinator/remoteControlBoard/icubSim/left_leg/rpc:o to /icubSim/left_leg/rp [... cut ...]
[INFO] |yarp.os.impl.PortCoreInputUnit|/walking-coordinator/remoteControlBoard/icubSim/right_leg/stateExt:i| Removing input from /icubSim/right_leg/stateExt:o to /walking-coordinator/remoteControlBoard/icubSim/right_leg/stateExt:i
[INFO] |yarp.os.RFModule| RFModule finished.
[WARNING] |yarp.os.NetworkClock| Destroying network clock
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP249/WalkingModule/8465/clock:i| Removing input from /clock to /IITICUBLAP249/WalkingModule/8465/clock:i
```
I didn't even issued a `setGoal` command.

mebbaid, 2022-02-16:
@S-Dafarra @GiulioRomualdi

GiulioRomualdi, 2022-02-16:
Hi @SimoneMic, the long story short is that I never spent a lot of time tuning the controller in simulation for iCubGazeboV3. For this reason, the set of gains may not be optimal. 
Here are some hints:
1. The contacts are not stable. It may happen that at the impact the contact is continuously activated and deactivated. This causes
   ```
   [ERROR] [evaluateZMP] None of the two contacts is valid.
   [ERROR] [WalkingModule::updateModule] Unable to evaluate the ZMP.
   ``` 
2. This error `[ERROR] [RobotInterface::setDirectPositionReferences] The worst error between the current and the desired position of the  l_shoulder_roll  joint is greater than 0.5 rad.` appears if the position tracking error of a joint, in this case, the `l_shoulder_roll`, is pretty high. This is due to the low-level position control.

lrapetti, 2022-02-16:
> > running the ankle joints in compliant mode
>     
>     
>     The compliant mode has been tested only on the real robot as a consequence I've never tuned the low-level gains for the compliant mode in simulation. Furthermore, the walking does not provide any interface to set them. If you want to change them you should open a pr in icub-models-generator

Concerning this point, all the parameters for complaint mode are actually set to zero in [`iCubGazeboV3`](https://github.com/robotology/icub-models/blob/ab11b8cb8799b85a8805be95e3d66e7c6c7585ff/iCub/conf_icub3/gazebo_icub_left_leg.ini#L64-L68) (and same for `iCubGazeboV2_5`).  Just as reference, we could try the parameters tuned on the real robot (that can be found [here](https://github.com/ami-iit/robots-configuration/blob/afbe7f7a580accd64c6cb5c23286275cf6df752b/iCubGenova09/hardware/motorControl/right_leg-eb12-j3_5-mc.xml#L27-L30))

mebbaid, 2022-03-17:
As per our last discussion, me and @SimoneMic  have arrived at a new set of gains for the low level position-control pid that works OK for the `stickBot` in both our machines. Additionally those gains seems to work as well, in my machine, for the `iCubGazeboV3`. These gains will update the `icub-model-generator` repo.

SimoneMic, 2022-03-25:
I think that we can close this issue.
With @mebbaid pid parameters there's a stable improvement and the simulations are more reliable. The robot still falls after a while, but lasts enough for testing localization algorithms and motion planning.
For mapping it could be another topic, but that could be a separate issue for future plans.

Thanks all!

closed 2022-03-25 by SimoneMic