gbionics/walking-controllers #109: Wrong ini configuraion file selected by the LoggerModule when running it wih iCubGazeboV3

nunoguedelha, 2022-03-09:
When we run the **LoggerModule** as described in [this documentation section](https://github.com/robotology/walking-controllers#how-to-dump-data), the logger opens the input port `/logger/data:i`. If we then run the **WalkingModule** as per the same documentation, and while setting `YARP_ROBOT_NAME = iCubGazeboV3`, the **WalkingModule** fails with the console output:
```
[ERROR] Unable to connect the ports  /walking-coordinator/logger/data:o and /yarp-robot-logger/exogenous_signals/walking:i
[INFO] |yarp.os.RFModule| RFModule failed to open.
Illegal instruction: 4
```
The **WalkingModule** fails because it's expecting the input port `/yarp-robot-logger/exogenous_signals/walking:i` instead of `/logger/data:i`.

nunoguedelha, 2022-03-09:
When we run the **LoggerModule** as described in [this documentation section](https://github.com/robotology/walking-controllers#how-to-dump-data), the default configuration file [dcmWalkingLogger.ini](https://github.com/robotology/walking-controllers/src/LoggerModule/app/dcmWalkingLogger/dcmWalkingLogger.ini) is used to initialize the module. For this reason, when running the **WalkingModule** with `YARP_ROBOT_NAME = iCubGazeboV3`, the logger input port opened by **LoggerModule** (`/logger/data:i`) doesn't match the one expected by the WalkingModule and defined in [src/WalkingModule/app/robots/iCubGazeboV3/dcm_walking/common/walkingLogger.ini](https://github.com/robotology/walking-controllers/src/WalkingModule/app/robots/iCubGazeboV3/dcm_walking/common/walkingLogger.ini) (`/yarp-robot-logger/exogenous_signals/walking:i`).

GiulioRomualdi, 2022-09-22:
Logger module has been removed after https://github.com/robotology/walking-controllers/pull/97

closed 2022-09-22 by GiulioRomualdi