gbionics/walking-controllers #122: Error during CMake Generate:

mlaudu, 2022-09-16:
Hi, I successfully configured the software in CMake with all dependencies found; but when I hit 'Configure', I got the following error message:

CMake Error at cmake/AddWalkingControllersApplication.cmake:30 (add_executable):
  Target "WalkingModule" links to target
  "BipedalLocomotion::VectorsCollection" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  src/WalkingModule/CMakeLists.txt:9 (add_walking_controllers_application)

Any advise will be greatly appreciated.

GiulioRomualdi, 2022-09-20:
Hi @mlaudu! Thank you for using walking-controllers! I've just noticed that the documentation has not been updated. I'm deeply sorry for that! 
You should install [`bipedal-locomotion-framework`](https://github.com/ami-iit/bipedal-locomotion-framework). The easiest way to install compile walking-controllers with all the dependencies is to use the [`robotology-superbuild`](https://github.com/robotology/robotology-superbuild) enabling the CMake option `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS`.

Another possibility is to install `walking-controllers` with conda as
```
conda install -c robotology walking-controllers
```

mlaudu, 2022-09-20:
Thank you very much Giulio. Using robotolgy-superbuild works well and the problem is resolved. Cheers.

mlaudu, 2022-09-20:
Issue resolved using robotolgy-superbuild.

closed 2022-09-20 by mlaudu

GiulioRomualdi, 2022-09-21:
You're welcome!