gbionics/walking-controllers #133: Cannot read from dcm_walking_with_joypad.ini

ykjia, 2022-12-09:
Hi!

I installed the walking-controllers and followed the "How to run the simulation" step-by-step. I am using the iCubGazeboV2_5 (no hands) model. When I run the walking module I have got some errors:

root@3da33a7aee3f:~# YARP_CLOCK=/clock WalkingModule
[INFO] |yarp.os.Port|/3da33a7aee3f/WalkingModule/2996/clock:i| Port /3da33a7aee3f/WalkingModule/2996/clock:i active at tcp://172.17.0.2:10103/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.impl.PortCoreInputUnit|/3da33a7aee3f/WalkingModule/2996/clock:i| Receiving input from /clock to /3da33a7aee3f/WalkingModule/2996/clock:i using tcp
[ERROR] |yarp.os.Property| cannot read from dcm_walking_with_joypad.ini
[ERROR] [getiDynTreeVectorFixSizeFromSearchable] Missing field  goal_port_scaling
[ERROR] [WalkingModule::configure] Failed while reading goal_port_scaling.
[INFO] |yarp.os.RFModule| RFModule failed to open.
[WARNING] |yarp.os.NetworkClock| Destroying network clock
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.impl.PortCoreInputUnit|/3da33a7aee3f/WalkingModule/2996/clock:i| Removing input from /clock to /3da33a7aee3f/WalkingModule/2996/clock:i

Could you give me some advice? Thank you very much.

S-Dafarra, 2022-12-09:
Hi @ykjia! Can you try running the following?
```
YARP_ROBOT_NAME=iCubGazeboV2_5 WalkingModule
```

If that works, I means that the ``YARP_ROBOT_NAME`` was not set, and you can simply add 
```
YARP_ROBOT_NAME=iCubGazeboV2_5
```
to your ``.bashrc`` to avoid adding it before ``WalkingModule``.

ykjia, 2022-12-12:
The issue is solved. Thank you!

closed 2022-12-12 by ykjia