gbionics/walking-controllers #138: Wrong Height Value of the root_link to World Transform from the FKsolver for ergoCubGazeboV1

SimoneMic, 2023-02-21:
## Expected Behavior
The `root_link` height in respect of the `world` frame should always match the height of the `root_link` with the sole in contact with the ground (i.e. `r_sole` or `l_sole`).

In stickBot for example:
[image]

## Actual Behavior
Premise: the robot walks and navigates correctly, this is only a strange deviation.

The transform of the Forward Kinematic solver compenetrates the ground, like in figure:
[image]

The calculations on the transform are expressed in the Code Snippet section.
Using the stickBot model in simulation we have the expected behavior.

Transform value:
```
- header:
    stamp:
      sec: 3653
      nanosec: 741000000
    frame_id: odom
  child_frame_id: root_link
  transform:
    translation:
      x: -0.03845819101269138
      y: -0.00029803977206156046
      z: 0.76088412317419
    rotation:
      x: 0.0015298974928377215
      y: -0.03812475686240274
      z: 0.00010598504987431123
      w: 0.9992718104174725
```

## Code snippet
In the `walking-controllers` `Module.cpp` the transform is exposed on a separate dedicated thread as follow:
```
root_linkTransform = m_FKSolver->getRootLinkToWorldTransform();

auto& transform = data.addList();
auto& data = m_unicyclePort.prepare();
transform.addFloat64(root_linkTransform.getPosition()(0));
transform.addFloat64(root_linkTransform.getPosition()(1));
transform.addFloat64(root_linkTransform.getPosition()(2));
transform.addFloat64(root_linkTransform.getRotation().asRPY()(0));
transform.addFloat64(root_linkTransform.getRotation().asRPY()(1));
transform.addFloat64(root_linkTransform.getRotation().asRPY()(2));
```
The ROS2 Node publishing the TF:
```
geometry_msgs::msg::TransformStamped odomTf;
odomTf.header.frame_id = "odom";
odomTf.child_frame_id = "root_link";
odomTf.header.stamp = tf.header.stamp;
odomTf.transform.translation.x = data.get(3).asList()->get(0).asFloat64();
odomTf.transform.translation.y = data.get(3).asList()->get(1).asFloat64();
odomTf.transform.translation.z = data.get(3).asList()->get(2).asFloat64();

tf2::Quaternion qOdom;
qOdom.setRPY(data.get(3).asList()->get(3).asFloat64(), data.get(3).asList()->get(4).asFloat64(), data.get(3).asList()->get(5).asFloat64());
odomTf.transform.rotation.x = qOdom.x();
odomTf.transform.rotation.y = qOdom.y();
odomTf.transform.rotation.z = qOdom.z();
odomTf.transform.rotation.w = qOdom.w();
tfBuffer.push_back(odomTf);
```

## Specifications
* I am using a custom walking-controller version for the unicycle planner, forked from master commit `1713405570bf6d1c0b6bc84568f266cd32882900`  `Date:   Wed Sep 28 19:06:25 2022 +0200`
* I am using the walking controller config files from the actual branch `ergoCubSN000` of `ergoCubGazeboV1` with the additional files for using `INVERSE_KINEMATICS_BLF_QP_SOLVER` from `ergoCubSN000`
* egoCub model from `ergocub-software` commit `3b629f1d566bc89bb6d81adcda9f4694237ab6`

##Conclusions
Since the navigation is in 2D, the impact of this situation should be minimal. However I am wondering if this could hinder the walking-controller stability/behavior.
It would be enough to know which files could impact this root_link height, so that I can check them.

Thanks!

S-Dafarra, 2023-02-21:
Hi @SimoneMic, could you share the link to the code? 

Also, I think I am not understanding the test you are running. The numeric value has a positive z (that is reasonable considering the world frame on the ground), but then why in the visualization it compenetrates the ground?

S-Dafarra, 2023-02-21:
Also, why you first add the list and then prepare ``data``?
```
auto& transform = data.addList();
auto& data = m_unicyclePort.prepare();
```
Shouldn't it be the opposite?

SimoneMic, 2023-02-21:
> Also, why you first add the list and then prepare `data`?
> 
> ```
> auto& transform = data.addList();
> auto& data = m_unicyclePort.prepare();
> ```
> 
> Shouldn't it be the opposite?

Yep, My bad copy paste, since it was other code in between.

SimoneMic, 2023-02-21:
> Hi @SimoneMic, could you share the link to the code?
> 
> Also, I think I am not understanding the test you are running. The numeric value has a positive z (that is reasonable considering the world frame on the ground), but then why in the visualization it compenetrates the ground?

Basically what I am doing is to have the odom frame in the ROS2 environment ([TF2](http://wiki.ros.org/tf2) to be specific) to use that for the robot localization with [AMCL](https://navigation.ros.org/configuration/packages/configuring-amcl.html).
In displaying the TF frames, I've noticed that there's an height mismatch, just this.

SimoneMic, 2023-02-21:
Meanwhile with stickBot this issue was not present

S-Dafarra, 2023-02-21:
> Basically what I am doing is to have the odom frame in the ROS2 environment ([TF2](http://wiki.ros.org/tf2) to be specific) to use that for the robot localization with [...]

Sorry I am lost, how did you notice this mismatch? The numeric data above seemed somehow reasonable

SimoneMic, 2023-02-21:
> Sorry I am lost, how did you notice this mismatch? The numeric data above seemed somehow reasonable

Actually I found an error in the robot_state_publisher in ROS2, was still using a wrong file path. So now it's solved. My bad.

S-Dafarra, 2023-02-21:
> Actually I found an error in the robot_state_publisher in ROS2, was still using a wrong file path. So now it's solved. My bad.

Thanks a lot for reporting any how :wink: