gbionics/walking-controllers #148: `ergocubSN000` walks mid-air, instead of stopping

SimoneMic, 2023-08-30:
Hello, 

While using `ergocubSN000` I had to pull it up due to a wrong path computed while navigating.
However, the walking-controller didn't stop, and continued "walking" in mid-air. There should be a check on the contacts of the feet that stops the controller but it didn't fired.
I did the calibration of the feet sensor before using the robot (while suspended on the crane), and also the preparation.

Here's the video: 

https://github.com/robotology/walking-controllers/assets/86918431/3928d4ca-e2bc-4e30-9254-e97347aed621

I am using my [fork](https://github.com/SimoneMic/walking-controllers/tree/ergoCub_SN000) but I haven't changed anything regarding the check on the contact feet.

DanielePucci, 2023-08-31:
CC @S-Dafarra @GiulioRomualdi @mebbaid

S-Dafarra, 2023-09-04:
Probably you need to increase this value: https://github.com/SimoneMic/walking-controllers/blob/ergoCub_SN000/src/WalkingModule/app/robots/ergoCubSN000/dcm_walking_with_joypad.ini#L26

Right now it is pretty low (1N). You can check what is the value of the normal force applied to the foot when the robot is hanging (you can sum the 3rd component of these two ports for example: https://github.com/robotology/robots-configuration/blob/edc514623e39330540aecb500d75ecac3874f7bc/ergoCubSN000/estimators/wholebodydynamics.xml#L48-L49)

Note that by increasing that value, the robot might stop more often while walking

SimoneMic, 2023-11-20:
Forgot to close this issue, sorry.

After updating the parameters it was working as intended, thanks

closed 2023-11-20 by SimoneMic