gbionics/walking-controllers #154: `prepareRobot` fails using ma97 as a solver

paolo-viceconte, 2023-11-08:
If I try to run the [walking simulation](https://github.com/robotology/walking-controllers#computer-how-to-run-the-simulation) using `ma97` as a solver, I cannot even prepare the robot.

The setup on my personal laptop (running Ubuntu 22.04) includes:
- robotology-superbuild [v2023.05.2](https://github.com/robotology/robotology-superbuild/releases/tag/v2023.05.2)
- HSL proprietary solvers installed through apt by following https://github.com/ami-iit/coinhsl-binary-packages#aptpip 

[Configuring the ik](https://github.com/robotology/walking-controllers#inverse-kinematics-configuration) with:
```
# solver paramenters
solver-verbosity        5
solver_name             ma97
max-cpu-time            20
```
I got:
```
$ yarp rpc /walking-coordinator/rpc
>> prepareRobot 
Response: [fail]
```
and on the WalkingModule side:
```
******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit http://projects.coin-or.org/Ipopt
******************************************************************************

This is Ipopt version 3.11.9, running with linear solver ma97.

Number of nonzeros in equality constraint Jacobian...:      126
Number of nonzeros in inequality constraint Jacobian.:       87
Number of nonzeros in Lagrangian Hessian.............:        0

Total number of variables............................:       29
                     variables with only lower bounds:        0
                variables with lower and upper bounds:       26
                     variables with only upper bounds:        0
Total number of equality constraints.................:       12
Total number of inequality constraints...............:        3
        inequality constraints with only lower bounds:        0
   inequality constraints with lower and upper bounds:        3
        inequality constraints with only upper bounds:        0

iter    objective    inf_pr   inf_du lg(mu)  ||d||  lg(rg) alpha_du alpha_pr  ls
   0  4.1623176e-04 1.74e-02 2.78e-02   0.0 0.00e+00    -  0.00e+00 0.00e+00   0
   1  3.0325037e-04 1.49e-02 5.80e+00  -5.9 6.29e-02    -  9.01e-01 1.41e-01h  1
   2  3.0679183e-04 1.47e-02 4.49e-01  -3.0 6.66e-02    -  9.58e-01 1.82e-02f  1
   3  2.2765575e-04 2.53e-02 2.42e+00  -3.1 1.70e+00    -  3.27e-01 6.01e-03h  7
WARNING: Problem in step computation; switching to emergency mode.
   4r 2.2765575e-04 2.53e-02 9.99e+02  -1.6 0.00e+00  20.0 0.00e+00 0.00e+00R  1
   5r 6.2874155e-04 6.42e-03 1.64e+02  -1.0 2.47e+01    -  1.00e+00 1.01e-03f  1
   6  5.1065758e-04 1.44e-02 6.72e-01  -3.3 5.09e-02    -  1.00e+00 1.25e-01h  4
WARNING: Problem in step computation; switching to emergency mode.
   7r 5.1065758e-04 1.44e-02 9.99e+02  -1.8 0.00e+00  20.0 0.00e+00 0.00e+00R  1
   8r 6.6152691e-04 3.51e-03 2.03e+02  -2.0 1.43e+01    -  1.00e+00 1.00e-03f  1
   9  6.6534377e-04 3.49e-03 2.29e+00  -4.3 1.18e-01    -  9.06e-01 1.95e-03h 10
WARNING: Problem in step computation; switching to emergency mode.
iter    objective    inf_pr   inf_du lg(mu)  ||d||  lg(rg) alpha_du alpha_pr  ls
  10r 6.6534377e-04 3.49e-03 9.99e+02  -2.5 0.00e+00  20.0 0.00e+00 0.00e+00R  1
  11r 1.3070540e-03 1.30e-03 3.66e+02  -2.7 1.43e+01    -  1.00e+00 9.94e-04f  1
  12  2.1179374e-02 2.57e-01 3.04e-01  -5.5 9.44e-02    -  8.70e-01 8.70e-01s 22
  13  1.7550089e-01 1.40e-01 7.45e-01  -5.2 1.38e+00    -  1.26e-01 3.61e-01h  1
  14  1.6529956e-01 1.83e-01 1.21e+00  -5.2 1.11e+04    -  5.35e-05 1.38e-06f  3
  15  8.7258074e-02 2.06e-01 1.32e+00  -5.2 4.24e+05    -  5.81e-08 5.57e-07f  1
WARNING: Problem in step computation; switching to emergency mode.
  16r 8.7258074e-02 2.06e-01 9.99e+02  -0.7 0.00e+00  20.0 0.00e+00 0.00e+00R  1
  17r 1.2460723e-01 4.92e-02 7.53e+02  -1.8 1.82e+02    -  1.00e+00 1.12e-03f  1
WARNING: Problem in step computation; switching to emergency mode.
  18r 1.2460723e-01 4.92e-02 9.99e+02  -1.3 0.00e+00  20.0 0.00e+00 0.00e+00R  1
  19r 1.4217412e-01 1.87e-02 5.47e+02  -1.1 9.92e+01    -  1.00e+00 1.06e-03f  1
[... 48 lines cut ...]

                                   (scaled)                 (unscaled)
Objective...............:   1.2226459282315812e-01    1.2226459282315812e-01
Dual infeasibility......:   1.6847026062211115e-01    1.6847026062211115e-01
Constraint violation....:   2.9163472418444092e-01    2.9163472418444092e-01
Complementarity.........:   6.4410788886282253e-02    6.4410788886282253e-02
Overall NLP error.......:   2.9163472418444092e-01    2.9163472418444092e-01

Number of objective function evaluations             = 136
Number of objective gradient evaluations             = 28
Number of equality constraint evaluations            = 136
Number of inequality constraint evaluations          = 136
Number of equality constraint Jacobian evaluations   = 51
Number of inequality constraint Jacobian evaluations = 51
Number of Lagrangian Hessian evaluations             = 0
Total CPU secs in IPOPT (w/o function evaluations)   =      0.078
Total CPU secs in NLP function evaluations           =      0.007

EXIT: Restoration Failed!
[ERROR] WalkingIK: Failed in finding a solution.
[ERROR] [WalkingModule::prepareRobot] Inverse Kinematics failed while computing the initial position.
```

:warning: Everything works fine instead using either `mumps` or `ma27` as solvers, but I would need `ma97` for other purposes (i.e. testing https://github.com/ami-iit/paper_romualdi_2022_icra_centroidal-mpc-walking).  
cc @GiulioRomualdi

traversaro, 2023-11-09:
I see you are using ipopt 3.11, just for understanding the problem, can you try this with a modern ipopt, either installed via coinbrew or switching to install deps via conda-forge?

traversaro, 2023-11-09:
Which version of coin-hsl are you using?

paolo-viceconte, 2023-11-09:
> I see you are using ipopt 3.11, just for understanding the problem, can you try this with a modern ipopt, either installed via coinbrew or switching to install deps via conda-forge?

I will try it by following part of this tutorial: https://github.com/ami-iit/ami-commons/blob/master/doc/casadi-ipopt-hsl.md.

> Which version of coin-hsl are you using?

`coinhsl-2019.05.21-ubuntu-22.04.deb` taken from https://github.com/ami-iit/coinhsl-binary-packages/releases/tag/v2019.05.21.1.

S-Dafarra, 2023-11-10:
I tried on my machine, where I had installed the ``hsl`` solvers from source and it is not failing:

```

******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit https://github.com/coin-or/Ipopt
******************************************************************************

This is Ipopt version 3.14.9, running with linear solver ma97.

Starting derivative checker for first derivatives.

* jac_g [    6,    8] = -3.8897845777292883e-04 v  ~ -3.8902214782865485e-04  [ 1.123e-04]
* jac_g [    6,   11] = -9.0722135866040318e-05 v  ~ -9.0771834493352799e-05  [ 4.970e-04]
* jac_g [    7,   11] = -2.1644753635755183e-04 v  ~ -2.1647128534141302e-04  [ 1.097e-04]
* jac_g [    8,   11] =  2.0573452283607647e-04 v  ~  2.0570212200254900e-04  [ 1.575e-04]
* jac_g [    7,   15] = -1.0895463521336159e-04 v  ~ -1.0897949209720537e-04  [ 2.281e-04]
* jac_g [    8,   25] =  1.6072936658013135e-04 v  ~  1.6067147612375265e-04  [ 3.603e-04]
* jac_g [    6,   27] = -1.8396421342292343e-04 v  ~ -1.8403056856186595e-04  [ 3.606e-04]
* jac_g [    8,   27] =  4.3234763305766317e-04 v  ~  4.3227643686805095e-04  [ 1.647e-04]
* jac_g [    6,   28] =  1.0648677687915709e-03 v  ~  1.0647482895365101e-03  [ 1.122e-04]
* jac_g [    7,   28] = -1.6015981858687929e-05 v  ~ -1.6031620475587260e-05  [ 1.564e-04]
* jac_g [    8,   28] =  5.0438480975912008e-04 v  ~  5.0430770670573111e-04  [ 1.529e-04]

Derivative checker detected 11 error(s).

Number of nonzeros in equality constraint Jacobian...:      126
Number of nonzeros in inequality constraint Jacobian.:       87
Number of nonzeros in Lagrangian Hessian.............:        0

Total number of variables............................:       29
                     variables with only lower bounds:        0
                variables with lower and upper bounds:       26
                     variables with only upper bounds:        0
Total number of equality constraints.................:       12
Total number of inequality constraints...............:        3
        inequality constraints with only lower bounds:        0
   inequality constraints with lower and upper bounds:        3
        inequality constraints with only upper bounds:        0

iter    objective    inf_pr   inf_du lg(mu)  ||d||  lg(rg) alpha_du alpha_pr  ls
   0  4.1623176e-04 1.74e-02 2.78e-02   0.0 0.00e+00    -  0.00e+00 0.00e+00   0
   1  3.0325037e-04 1.49e-02 5.80e+00  -5.9 6.29e-02    -  9.01e-01 1.41e-01h  1
   2  3.0679183e-04 1.47e-02 4.49e-01  -3.0 6.66e-02    -  9.58e-01 1.82e-02f  1
   3  3.1049014e-03 8.31e-04 3.66e-02  -3.6 5.60e-02    -  1.00e+00 1.00e+00h  1
   4  2.5236479e-03 1.87e-05 2.55e-02  -4.5 3.12e-02    -  1.00e+00 1.00e+00h  1
   5  2.2917329e-03 8.56e-07 6.60e-03  -5.3 7.99e-03    -  1.00e+00 1.00e+00h  1
   6  2.2781252e-03 8.36e-08 4.77e-04  -6.2 8.26e-04    -  1.00e+00 1.00e+00h  1
   7  2.2762725e-03 8.84e-10 7.98e-05  -7.5 1.71e-04    -  1.00e+00 1.00e+00h  1

Number of Iterations....: 7

                                   (scaled)                 (unscaled)
Objective...............:   2.2762725261595727e-03    2.2762725261595727e-03
Dual infeasibility......:   7.9784028649363759e-05    7.9784028649363759e-05
Constraint violation....:   8.8358215882822232e-10    8.8358215882822232e-10
Variable bound violation:   0.0000000000000000e+00    0.0000000000000000e+00
Complementarity.........:   1.6100627884067756e-07    1.6100627884067756e-07
Overall NLP error.......:   7.9784028649363759e-05    7.9784028649363759e-05

Number of objective function evaluations             = 8
Number of objective gradient evaluations             = 8
Number of equality constraint evaluations            = 8
Number of inequality constraint evaluations          = 8
Number of equality constraint Jacobian evaluations   = 8
Number of inequality constraint Jacobian evaluations = 8
Number of Lagrangian Hessian evaluations             = 0
Total seconds in IPOPT                               = 0.575

EXIT: Optimal Solution Found.
q desired IK     8.24601   0.0573576 -0.00712427    -3.32021     19.2222    -12.8797     40.5267    -3.31471     19.2197    -12.8793      40.524     5.32928    0.876454    -0.21579    -35.0319    -21.3093   -0.920483     5.34256    0.900903   -0.218481    -35.0613    -21.3255   -0.909155
```
(yet it finds some possible errors in the derivatives)

traversaro, 2023-11-10:
@akhilsathuluri also had problems with ma97, even if in another context.

traversaro, 2023-11-10:
Something else that I think it could be worth is try a recent version of coinhsl/ma27, what we are using is a 4 years old version.

akhilsathuluri, 2023-11-14:
For me, installing `coin-hsl` from here worked: https://github.com/ami-iit/coinhsl-binary-packages
Previously, @CarlottaSartore also had issues with `ma97`

GiulioRomualdi, 2024-04-08:
https://github.com/robotology/walking-controllers/pull/159 makes mumps the linear solver used for `preparing` the robot. I'm going to close this issue

closed 2024-04-08 by GiulioRomualdi