gbionics/walking-controllers #185: WalkingModule segmentation fault at prepareRobot

giotherobot, 2024-08-13:
I just updated the superbuild with a freshly created conda environment

<details><summary>Conda list</summary>
<p>

```
# packages in environment at /home/giovannif/mambaforge/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_16    conda-forge
ace                       8.0.1                he02047a_0    conda-forge
aiohappyeyeballs          2.3.5              pyhd8ed1ab_0    conda-forge
aiohttp                   3.10.3          py312h41a817b_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.12               h4ab18f5_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.9.1                hac33072_0    conda-forge
asio                      1.29.0               h59595ed_0    conda-forge
assimp                    5.3.1                h8343317_3    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h04ea711_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     24.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.20               h5f1c8d9_0    conda-forge
aws-c-cal                 0.6.12               h2ba76a8_0    conda-forge
aws-c-common              0.9.17               h4ab18f5_0    conda-forge
aws-c-compression         0.2.18               h36a0aea_4    conda-forge
aws-c-event-stream        0.4.2               h161de36_10    conda-forge
aws-c-http                0.8.1               h63f54a0_13    conda-forge
aws-c-io                  0.14.8               h96d4d28_0    conda-forge
aws-c-mqtt                0.10.4               hcc7299c_2    conda-forge
aws-c-s3                  0.5.9                h10bd90f_0    conda-forge
aws-c-sdkutils            0.1.16               h36a0aea_0    conda-forge
aws-checksums             0.1.18               h36a0aea_4    conda-forge
aws-crt-cpp               0.26.8              h4f3a3cc_11    conda-forge
aws-sdk-cpp               1.11.267             h51dfee4_8    conda-forge
azure-core-cpp            1.11.1               h91d86a7_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_1    conda-forge
azure-storage-common-cpp  12.5.0               h94269e2_4    conda-forge
bash-completion           2.11                 ha770c72_1    conda-forge
binutils                  2.40                 h4852527_7    conda-forge
binutils_impl_linux-64    2.40                 ha1999f0_7    conda-forge
binutils_linux-64         2.40                 hb3c18ed_0    conda-forge
blosc                     1.21.6               hef167b5_0    conda-forge
boost                     1.84.0               h9cebb41_3    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.25                 hfb8ada1_2    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.33.0               ha66036c_0    conda-forge
c-compiler                1.7.0                hd590300_1    conda-forge
ca-certificates           2024.7.4             hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
certifi                   2024.7.4           pyhd8ed1ab_0    conda-forge
cfitsio                   4.4.0                hbdc6101_1    conda-forge
cli11                     2.4.1                h59595ed_0    conda-forge
cmake                     3.30.2               hf8c4bd3_0    conda-forge
compilers                 1.7.0                ha770c72_1    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.2.1           py312h8572e83_0    conda-forge
cppzmq                    4.10.0               h2e2a08d_1    conda-forge
cudatoolkit               11.8.0              h4ba93d1_13    conda-forge
cxx-compiler              1.7.0                h00ab1b0_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.2               h89be5ba_2    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
[... 389 lines cut ...]
xorg-libxxf86vm           1.1.5                h4bc722e_1    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-recordproto          1.14.2            h7f98852_1002    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-util-macros          1.19.3               h7f98852_0    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.4           py312h98912ed_0    conda-forge
zeromq                    4.3.5                h75354e8_4    conda-forge
zlib                      1.3.1                h4ab18f5_1    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
zziplib                   0.13.69              he45264a_2    conda-forge
```

</p>
</details> 

And now I am trying to run my usual simulation environment with gazebo, but the WalkingModule crashes with a segmentation fault when I try to prepareRobot in this portion of the code:

https://github.com/robotology/idyntree/blob/94f051df561a91546f9e245fc925e2fb3691aa99/src/inverse-kinematics/src/InverseKinematicsNLP.cpp#L1346C1-L1346C32

(I am not sure whether to report this here or in IDynTree now. )

traversaro, 2024-08-13:
If you changed the conda env, you need to recompile the whole superbuild. In theory the compiler should be smart enough to recompile everything accounting for ABI changes, but sometimes there is some tricky corner case that could lead to some ABI conflicts. Just to be safe, it is a good idea to recompile the superbuild from scratch if you changed a lot the conda environment.

giotherobot, 2024-08-14:
Rebuilding from scratch solved the issue.

closed 2024-08-14 by giotherobot