gbionics/walking-controllers #195: setGoal not working

isorrentino, 2025-01-17:
The modifications introduced with this commit https://github.com/robotology/walking-controllers/commit/f9c8e93cf39bb0c679fe2d2819ea5dc4ee584ce9, caused failure of setGoal command form rpc port.
Even if I get OK after launching the command, the robot does not walk.

S-Dafarra, 2025-01-17:
It is possible that we would have to send commands repeatedly. We would need to understand how to differentiate inputs arriving from RPC, and not from port.

GiulioRomualdi, 2025-01-20:
Should we remove setGoal from terminal?

S-Dafarra, 2025-01-20:
This could also be an option indeed. Maybe we can change the documentation to use the keyboard joypad?

isorrentino, 2025-01-20:
> This could also be an option indeed. Maybe we can change the documentation to use the keyboard joypad?

It is a good idea in my opinion.

GiulioRomualdi, 2025-01-23:
I think this will also simplify the state machine in the walking