gbionics/walking-controllers #201: WalkingJoypadModule does not start

GiulioRomualdi, 2025-08-05:
I try try run WalkingJoypadModule with the keyboard option but I get a glew error

```
WalkingJoypadModule --from keyboard.ini
[DEBUG] |yarp.dev.PolyDriver|keyboardJoypad| Parameters are JOYPAD_DEVICE (device keyboardJoypad) (axes (ad ws left_right)) (wasd_label "Forward and rotation") (arrows_label "Left and right translation") (buttons ("space-J0:Prepare" "enter-J1:Start" "delete-J2:Stop" "p-J3:Pause" "n-J4:GoalConn" "m-J5:RPCConn" "k-J7:Disconnect")) (button_size 180.0) (font_multiplier 1.30000000000000004441) (window_width 1920) (window_height 1010) (no_gui_thread true) (joypad_deadzone 0.200000000000000011102) (padding 50)
[INFO] |yarp.device.keyboard.joypad| The key min_button_size is not present in the configuration file. Using the default value: 50
[INFO] |yarp.device.keyboard.joypad| The key max_button_size is not present in the configuration file. Using the default value: 200
[INFO] |yarp.device.keyboard.joypad| The key min_font_multiplier is not present in the configuration file. Using the default value: 0.5
[INFO] |yarp.device.keyboard.joypad| The key max_font_multiplier is not present in the configuration file. Using the default value: 4
[INFO] |yarp.device.keyboard.joypad| The key gui_period is not present in the configuration file. Using the default value: 0.033
[INFO] |yarp.device.keyboard.joypad| The key buttons_per_row is not present in the configuration file. Using the default value: 3
[INFO] |yarp.device.keyboard.joypad| The key "allow_window_closing" is not present in the configuration file. Using the default value: false
[INFO] |yarp.device.keyboard.joypad| The key "joypads_index" is not present in the configuration file. Using only the joypad with index 0 (if present).
[INFO] |yarp.device.keyboard.joypad| The key ad_joypad_axis_index is not present in the configuration file. Using the default value: 0
[INFO] |yarp.device.keyboard.joypad| The key ws_joypad_axis_index is not present in the configuration file. Using the default value: 1
[INFO] |yarp.device.keyboard.joypad| The key left_right_joypad_axis_index is not present in the configuration file. Using the default value: 2
[INFO] |yarp.device.keyboard.joypad| The key up_down_joypad_axis_index is not present in the configuration file. Using the default value: 3
[INFO] |yarp.device.keyboard.joypad| The device is running in single threaded mode.
[INFO] |yarp.dev.PolyDriver|keyboardJoypad| Created device <keyboardJoypad>. See C++ class yarp::dev::KeyboardJoypad for documentation.
[INFO] |yarp.os.Port|/joypad/rpc| Port /joypad/rpc active at tcp://10.0.0.137:10002/
[INFO] |yarp.os.Port|/joypad/goal:o| Port /joypad/goal:o active at tcp://10.0.0.137:10003/
[ERROR] |yarp.device.keyboard.joypad| glewInit failed, aborting.
[ERROR] |yarp.device.keyboard.joypad| glewInit failed, aborting.
[ERROR] |yarp.device.keyboard.joypad| glewInit failed, aborting.
[ERROR] |yarp.device.keyboard.joypad| glewInit failed, aborting.
[ERROR] |yarp.device.keyboard.joypad| glewInit failed, aborting.

```

I installed all the deps with conda 
```
# packages in environment at /home/gromualdi-iit.local/miniforge3/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       8.0.2                hc8b94ec_0    conda-forge
adwaita-icon-theme        48.0                     unix_0    conda-forge
aiohappyeyeballs          2.6.1              pyhd8ed1ab_0    conda-forge
aiohttp                   3.11.18          py39h9399b63_0    conda-forge
aiosignal                 1.3.2              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.14               hb9d3cd8_0    conda-forge
ampl-asl                  1.0.0                h5888daf_2    conda-forge
aom                       3.9.1                hac33072_0    conda-forge
asio                      1.29.0               h59595ed_0    conda-forge
assimp                    5.4.3                h8943939_0    conda-forge
asttokens                 3.0.0              pyhd8ed1ab_1    conda-forge
async-timeout             5.0.1              pyhd8ed1ab_1    conda-forge
at-spi2-atk               2.38.0               h0630a04_3    conda-forge
at-spi2-core              2.40.3               h0630a04_0    conda-forge
atk-1.0                   2.38.0               h04ea711_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     25.3.0             pyh71513ae_0    conda-forge
aws-c-auth                0.9.0                h59ae206_7    conda-forge
aws-c-cal                 0.9.0                h5e3027f_1    conda-forge
aws-c-common              0.12.3               hb9d3cd8_0    conda-forge
aws-c-compression         0.3.1                hafb2847_5    conda-forge
aws-c-event-stream        0.5.4                h2dcaabb_9    conda-forge
aws-c-http                0.10.1               hb50fa74_1    conda-forge
aws-c-io                  0.19.0               h7962f60_2    conda-forge
aws-c-mqtt                0.13.0               h35de22e_3    conda-forge
aws-c-s3                  0.7.17               h50d7d24_2    conda-forge
aws-c-sdkutils            0.2.3                hafb2847_5    conda-forge
aws-checksums             0.2.7                hafb2847_1    conda-forge
aws-crt-cpp               0.32.5               h2811929_3    conda-forge
aws-sdk-cpp               1.11.510             hffe9a0f_8    conda-forge
azure-core-cpp            1.14.0               h5cfcd09_0    conda-forge
azure-identity-cpp        1.10.0               h113e628_0    conda-forge
azure-storage-blobs-cpp   12.13.0              h3cf044e_1    conda-forge
azure-storage-common-cpp  12.8.0               h736e048_1    conda-forge
[... 462 lines cut ...]
xorg-libxi                1.8.2                hb9d3cd8_0    conda-forge
xorg-libxinerama          1.1.5                h5888daf_1    conda-forge
xorg-libxmu               1.2.1                hb9d3cd8_1    conda-forge
xorg-libxpm               3.5.17               hb9d3cd8_1    conda-forge
xorg-libxrandr            1.5.4                hb9d3cd8_0    conda-forge
xorg-libxrender           0.9.12               hb9d3cd8_0    conda-forge
xorg-libxscrnsaver        1.2.4                hb9d3cd8_0    conda-forge
xorg-libxshmfence         1.3.3                hb9d3cd8_0    conda-forge
xorg-libxt                1.3.1                hb9d3cd8_0    conda-forge
xorg-libxtst              1.2.5                hb9d3cd8_3    conda-forge
xorg-libxxf86vm           1.1.6                hb9d3cd8_0    conda-forge
xorg-xf86vidmodeproto     2.3.1             hb9d3cd8_1005    conda-forge
xorg-xorgproto            2024.1               hb9d3cd8_1    conda-forge
xz                        5.8.1                hbcc6ac9_1    conda-forge
xz-gpl-tools              5.8.1                hbcc6ac9_1    conda-forge
xz-tools                  5.8.1                hb9d3cd8_1    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.20.0           py39h9399b63_0    conda-forge
zeromq                    4.3.5                h3b0a872_7    conda-forge
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zlib                      1.3.1                hb9d3cd8_2    conda-forge
zstd                      1.5.7                hb8e6e7a_2    conda-forge
zziplib                   0.13.69              he45264a_2    conda-forge

```

S-Dafarra, 2025-08-05:
On my Windows side it works with the following env
```
(robsub) D:\>conda list
# packages in environment at D:\miniforge3\envs\robsub:
#
# Name                    Version                   Build  Channel
ace                       8.0.1                he0c23c2_0    conda-forge
aiohappyeyeballs          2.4.0              pyhd8ed1ab_0    conda-forge
aiohttp                   3.10.5          py312h4389bb4_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.9.1                he0c23c2_0    conda-forge
asio                      1.29.0               h63175ca_0    conda-forge
assimp                    5.3.1                h0dbab56_3    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
attrs                     24.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.20               h6823eb1_0    conda-forge
aws-c-cal                 0.6.12               hc83774a_0    conda-forge
aws-c-common              0.9.17               h2466b09_0    conda-forge
aws-c-compression         0.2.18               hc83774a_4    conda-forge
aws-c-event-stream        0.4.2               hc6c0aac_10    conda-forge
aws-c-http                0.8.1               hced5053_13    conda-forge
aws-c-io                  0.14.8               hebaacdb_0    conda-forge
aws-c-mqtt                0.10.4               hdafd9a4_2    conda-forge
aws-c-s3                  0.5.9                h7a83f0e_0    conda-forge
aws-c-sdkutils            0.1.16               hc83774a_0    conda-forge
aws-checksums             0.1.18               hc83774a_4    conda-forge
aws-crt-cpp               0.26.8              h672a689_11    conda-forge
aws-sdk-cpp               1.11.267             h12f3f85_8    conda-forge
azure-core-cpp            1.11.1               h249a519_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h91493d7_1    conda-forge
azure-storage-common-cpp  12.5.0               h91493d7_4    conda-forge
blosc                     1.21.6               h85f69ea_0    conda-forge
boost                     1.84.0               h7e22eef_5    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
bullet-cpp                3.25                 h2ab9e98_2    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
c-ares                    1.33.1               h2466b09_0    conda-forge
c-compiler                1.7.0                hcfcfb64_1    conda-forge
ca-certificates           2024.12.14           h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h91e5215_2    conda-forge
catch2                    3.7.1                hc790b64_0    conda-forge
certifi                   2024.12.14         pyhd8ed1ab_0    conda-forge
cfitsio                   4.4.0                h9b0cee5_1    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cli11                     2.4.1                h63175ca_0    conda-forge
cmake                     3.30.2               h400e5d1_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.7.0                h57928b3_1    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.3.0           py312hd5eb7cc_0    conda-forge
cppzmq                    4.10.0               h42135b4_1    conda-forge
cudatoolkit               11.8.0              h09e9e62_13    conda-forge
cxx-compiler              1.7.0                h91493d7_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.2               h15b4ae7_2    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.4.1                h63175ca_0    conda-forge
double-conversion         3.3.0                h63175ca_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
exceptiongroup            1.2.2              pyhd8ed1ab_0    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
expat                     2.6.2                h63175ca_0    conda-forge
fcl                       0.7.0                he22821c_4    conda-forge
ffmpeg                    6.1.2           gpl_h9cf63cc_102    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
[... 307 lines cut ...]
wslink                    2.1.2              pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.5                he0c23c2_1    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.1.1                hcd874cb_0    conda-forge
xorg-libsm                1.2.4                hcd874cb_0    conda-forge
xorg-libx11               1.8.9                h0076a8d_1    conda-forge
[... cut ...]
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yarl                      1.9.4           py312he70551f_0    conda-forge
zeromq                    4.3.5                he1f189c_4    conda-forge
zlib                      1.3.1                h2466b09_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
zziplib                   0.13.69              h3ca93ac_2    conda-forge
```

Can you try downgrading ``sdl2`` to 2.30.5?

S-Dafarra, 2025-08-05:
Following https://github.com/nigels-com/glew/issues/417#issuecomment-2675406210 you can also edit https://github.com/ami-iit/yarp-device-keyboard-joypad/blob/4d182141d71b71c938d118533273337bfebf2f7e/src/devices/keyboard-joypad/KeyboardJoypad.cpp#L816 allowing also the value "4"

GiulioRomualdi, 2025-08-05:
Switching to x fixed the problem :()